ros-perception/image_pipeline #114: error on ubuntu 14.04. with ROS indigo

gitunit, 2015-01-18:
whenever i try to use image_view (mono or stereo) it simply doesn't work (at least for me)

for the following command:
**rosrun image_view image_view image:=/image_raw**

i get this error:
[ INFO] [1421603126.734295964]: Using transport "raw"
init done 
opengl support available 

(image_view:6927): GLib-GObject-WARNING **: invalid uninstantiatable type '(null)' in cast to 'GtkWidget'

(image_view:6927): GLib-GObject-WARNING **: instance of invalid non-instantiatable type '(null)'

(image_view:6927): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

vrabaud, 2015-01-18:
which version of Ubuntu, ROS and image_pipeline ?

gitunit, 2015-01-18:
(as in the headline mentioned) i use ubuntu 14.04., ROS Indigo and version of image_pipeline is 1.12.12 (the latest i guess)

weblucas, 2015-01-19:
I am with the same problem, on my laptop and on my Desktop. same, latest ubuntu 14.04, Ros indigo via apt-get. I also get the same error with disparity_view. my gtk versions are

ii  libgtk-3-0:amd64                                      3.10.8-0ubuntu1.4                                   amd64        
ii  libgtk-3-bin                                          3.10.8-0ubuntu1.4                                   amd64        
ii  libgtk-3-common                                       3.10.8-0ubuntu1.4                                   all          
ii  libgtk2.0-0:amd64                                     2.24.23-0ubuntu1.1                                  amd64       
ii  libgtk2.0-bin                                         2.24.23-0ubuntu1.1                                  amd64        
ii  libgtk2.0-common                                      2.24.23-0ubuntu1.1                                  all          
ii  libgtk2.0-dev                                         2.24.23-0ubuntu1.1                                  amd64        

thank

gitunit, 2015-01-31:
vrabaud, what do you think, should i downgrade the image_pipeline?

gitunit, 2015-02-06:
ok, got it. version 1.12.10 works. tried only image_view so far.

don't know what's about the 
**QMetaMethod::invoke: Unable to invoke methods with return values in queued connections** message though.

closed 2015-02-08 by gitunit

kanirudh54, 2015-06-10:
I still face the same problem described above.  Version 1.12.10 worked for me but had [issues](https://github.com/l0g1x/DUO-Camera-ROS/issues/18) with other packages.