ros-perception/image_pipeline #116: Stereo camera calibration doesn't display results

fariner, 2015-03-06:
Hello,

i can calibrate my stereo camera and commit but i cant see the result of the calibration in the top right of the application. After push in the calibrate buttom i see the following in the top right:

epi.
?

Why i cant see the results?

I execute the calibration with:
rosrun camera_calibration cameracalibrator.py --size 8x6 --square 0.024 right:=/stereo/right/image_raw left:=/stereo/left/image_raw right_camera:=/stereo/right left_camera:=/stereo/left  approximate:=5

(I have the checkerboard 8x6 in a sheet of paper)

Best Regards.

vrabaud, 2015-03-08:
which version do you have installed ? Which Ubuntu version ?
Do you have a rosbag I could use to test ?

fariner, 2015-03-11:
Hello,

sorry for the delay.

-I have Ubuntu 14.04 and ROS Indigo.

-I have created a Rosbag.
https://mega.co.nz/#!3YpBUJbC!OXCq5SA4wjFAQzigovenjlDW4Wf6BFLUxyPRL4wn0BA

-I run the rosbag record with:
rosbag record /stereo/left/camera_info /stereo/left/image_raw /stereo/right/camera_info /stereo/right/image_raw

-I run the camera calibration with:
rosrun camera_calibration cameracalibrator.py --size 8x6 --square 0.0248 right:=/stereo/right/image_raw left:=/stereo/left/image_raw left_camera:=/left right_camera:=/right --approximate=0.01

Thank you.

Best regards.

fariner, 2015-03-17:
Hello,

can you to help me?

Thank you in advance.

Best regards.

JWhitleyAStuff, 2019-04-25:
Closing very old thread.

closed 2019-04-25 by JWhitleyAStuff