ros-perception/image_pipeline #117: image_view doesn't work on Indigo?

End-Effector, 2015-03-13:
I'm just try this pkg out I use this command: rosrun image_view image_view image:=/camera/image

I get all this errors:

[ INFO] [1426247696.506904895]: Using transport "raw"
init done 
opengl support available 

(image_view:11333): GLib-GObject-WARNING **: invalid uninstantiatable type '(null)' in cast to 'GtkWidget'

(image_view:11333): GLib-GObject-WARNING **: instance of invalid non-instantiatable type '(null)'

(image_view:11333): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

vrabaud, 2015-03-13:
do you have the latest version of the package ? 1.12.12
Which version do you have ?
Does that happen every time ?
Are you on Ubuntu ? Which version ? Thx

End-Effector, 2015-03-13:
So I'm on Ubuntu 14.04 LTS
It happens everytime, and then crashes/blocks the command window.
Yes I'm on 1.12.12
Thank you.

vrabaud, 2015-03-13:
any chance you would have a rosbag with a few images ? Thx

End-Effector, 2015-03-14:
Can you please explain what You want me to do with the rosbag? You want pictures of the error happening or want me to use image view with a rosbag?

This error happens just by runing the image_view node, even with no messages on the image view topic.

vrabaud, 2015-03-14:
ok, I cannot reproduce that ....
Can you please try with older versions of image_view, especially 1.12.5 to see if that works for you ? Thx.

vrabaud, 2015-03-14:
did you compile OpenCV on your own ? Do you have some OpenCV in /usr/lib/local. I do not get "init done" and "opengl support available " (which are OpenCV messages but not on my default Ubuntu).

End-Effector, 2015-03-15:
Yes I have OpenCV installed on my own with youtube help yes. Maybe I mess that up. I followed this tutotial https://www.youtube.com/watch?v=ubOa_tJDM_U. But OpenCV is actually working well, I'm using it to replace image_view, to watch my ROS images/videos. I'm going to try to see if the 1.12.5 works.

End-Effector, 2015-03-15:
Hello,

 same problem with 1.12.5 just tried it.

paulo@paulo-Satellite-A660-1CJ:~/catkin_ws$ rosrun image_view image_view imagime:=/ar_single_board/result
[ WARN] [1426419853.171839430]: Topic 'image' has not been remapped! Typical command-line usage:
	$ rosrun image_view image_view image:=<image topic> [transport]
init done 
opengl support available 

(image_view:7996): GLib-GObject-WARNING **: invalid uninstantiatable type '(null)' in cast to 'GtkWidget'

(image_view:7996): GLib-GObject-WARNING **: instance of invalid non-instantiatable type '(null)'

(image_view:7996): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

vrabaud, 2015-03-15:
ok, that makes sense then. Which version of OpenCV do you have ? I believe some graphic threading issues got fixed, please update to the latest.

lilouch, 2015-04-07:
I have the same error. How i can update the package if you fixed the error  ? Thank.

bit-pirate, 2015-04-07:
Same error here. No window opens and program seems to freeze. Image source is a the Gazebo Kinect plugin.

```
$ rosrun image_view image_view image:=/gopher/sensor_3d_short_range/rgb/image_raw
[ INFO] [1428408004.395125149]: Using transport "raw"
init done 
opengl support available 

(process:27925): GLib-GObject-WARNING **: invalid uninstantiatable type '(null)' in cast to 'GtkWidget'

(process:27925): GLib-GObject-WARNING **: instance of invalid non-instantiatable type '(null)'

(process:27925): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
```

Running: Ubuntu 14.04 LTS, 64 bit, ROS Indigo, image_view 1.12.12, OpenCV version 2.4.8

vrabaud, 2015-04-14:
can you please try with 1.12.13 in shadow-fixed ? thx

lilouch, 2015-04-15:
I already downloaded it but how i can update the old version ? It seems it still doesn't work for me.
Whtat do you mean by shadow fixed ?

vrabaud, 2015-04-15:
Either compile it from trunk, either you add shadow-fixed as a repo (http://wiki.ros.org/ShadowRepository)
Either you manually download it from the repo: http://packages.ros.org/ros-shadow-fixed/ubuntu/pool/main/r/ros-indigo-image-view/

bit-pirate, 2015-04-17:
Checked out the source, compiled and tested it. Results are the same. No window pops up.

vrabaud, 2015-04-18:
@bit-pirate , @lilouch : do you have your own OpenCV installed ? The opengl message means it's not the Ubuntu package

lilouch, 2015-04-22:
Okey I've followed the instruction there http://wiki.ros.org/ShadowRepository and then used sudo apt-get install ros-indigo-image-view
I still have the error message
[ WARN] [1429761181.180650859]: Topic 'image' has not been remapped! Typical command-line usage:
	$ rosrun image_view image_view image:=<image topic> [transport]
[ INFO] [1429761181.281969051]: Using transport "raw"
init done 
opengl support available 

(image_view:6867): GLib-GObject-WARNING **: invalid uninstantiatable type '(null)' in cast to 'GtkWidget'

(image_view:6867): GLib-GObject-WARNING **: instance of invalid non-instantiatable type '(null)'

(image_view:6867): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

@vrabaud what do you mean by your own OpenCV ? Yes I installed it from a script which has downloaded and installed the latest one (2.4.11)

bit-pirate, 2015-04-22:
I have installed:

 * ros-indigo-cv-backports 0.1.3
 * ros-indigo-cv-bridge 1.11.7
 * ros-indigo-opencv 2.4.16
 * ros-indigo-opencv-candidate  0.2.4
 * libopencv-* 2.4.8

Btw, I'm using KDE (Kubuntu).

dgbwolf, 2015-04-23:
I had the same issue with opencv 2.4.9 installed. I completely removed ros and opencv and after that reinstalled ros. Now it shows opencv 2.4.8 as installed and rosrun image_view ... works.

vrabaud, 2015-04-23:
@dgbwolf : this is due to your OpenCV that was compiled with OpenGL support. OpenCV 2.4.10 has fixed for all those threading issues and it's probably time we build image_view on cv_backports.
rqt_image_view also works very well.

kschwan, 2015-05-04:
In OpenCV 2.4.11 built against Qt, the cvStartWindowThread function is simply implemented as

    CV_IMPL int cvStartWindowThread()
    {
        return 0;
    }

I suppose the user is then responsible for creating windows and processing ROS callbacks in separate threads.

udit7395, 2015-06-19:
Similar error here rqt_image_view works well with uncompressed images but get segmentation fault when using compressed images {from my android camera}. 
http://stackoverflow.com/questions/30925252/imageview-giving-glib-error

udit7395, 2015-07-03:
Rebuiding OpenCV with Qt-4 or Qt==off .fixes the problem

mihaibujanca, 2015-08-20:
@udit7395 Rebuilt opencv with QT off and still doesn't work for me, also upgraded now to opencv 3.0 (from 2.4.9) and the problem persists.

vrabaud, 2015-08-20:
@udit7395 , is the error the same or not ? does rqt_image_view work for you ?

udit7395, 2015-08-20:
@vrabaud rqt_image_view did work for me using my standard laptop webcam but when i used to see images from my android camera through cv_bridge it crashed .
@mihaibujanca I am using OpenCV 2.4.10 .

vrabaud, 2015-12-03:
#166 should fix all the threading issues.

closed 2015-12-03 by vrabaud

bosslegend33, 2016-02-27:
@vrabaud hello. I'm not that experience with ROS but I am having the same issue while launching the image_view node in ros-indigo and with opencv-2.4.10. If I reinstall the ros-indigo-image* packages with sudo ap-get will this issue be fixed or I should reinstall them from the github source? Thanks in advance.

wkentaro, 2016-02-27:
マムマヌカンん買う(￣^￣)ゞスサスは差

wkentaro, 2016-02-27:
sorry, it's mistyping on my iPhone.

vrabaud, 2016-02-28:
@bosslegend33 , you have a custom OpenCV right ? Which version did you install ? This is not supported. Also, which version of image_view do you have ? Please install from source and see if that works for you. Thx.

bosslegend33, 2016-02-28:
@vrabaud I installed opencv 2.4.10 in ubuntu 14.04 with ros indigo. Unfortunately, I need some of the nonfree modules of opencv for a project. I have the version 1.12.15 of image_view. I will try and inform. Thanks.