ros-perception/image_pipeline #119: camera_info from decimate is incorrect

cbames, 2015-03-18:
After using the software decimation to reduce the size of an image I'm using I noticed that besides binning_x and binning_y, the intrinsics have not been updated to reflect the new resolution of the image. 

Here's the incoming message: 
```
header: 
  seq: 225
  stamp: 
    secs: 1426715141
    nsecs: 857637301
  frame_id: r2/head/prosilica/left_optical_frame
height: 2050
width: 2448
distortion_model: plumb_bob
D: [-0.12297000000000001, 0.13763, -0.0008300000000000001, -0.00303, -0.08479]
K: [2348.02633, 0.0, 1173.59324, 0.0, 2346.76611, 994.74302, 0.0, 0.0, 1.0]
R: [0.99916, 0.011640000000000001, -0.03925, -0.01191, 0.9999100000000001, -0.006730000000000001, 0.03917, 0.00719, 0.99921]
P: [2454.07, 0.0, 1218.16699, 0.0, 0.0, 2454.07, 991.63319, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 1
binning_y: 1
roi: 
  x_offset: 0
  y_offset: 0
  height: 2050
  width: 2448
  do_rectify: False
```
and the outbound message: 
```
header: 
  seq: 13
  stamp: 
    secs: 1426715393
    nsecs: 806356956
  frame_id: r2/head/prosilica/left_optical_frame
height: 2050
width: 2448
distortion_model: plumb_bob
D: [-0.12297000000000001, 0.13763, -0.0008300000000000001, -0.00303, -0.08479]
K: [2348.02633, 0.0, 1173.59324, 0.0, 2346.76611, 994.74302, 0.0, 0.0, 1.0]
R: [0.99916, 0.011640000000000001, -0.03925, -0.01191, 0.9999100000000001, -0.006730000000000001, 0.03917, 0.00719, 0.99921]
P: [2454.07, 0.0, 1218.16699, 0.0, 0.0, 2454.07, 991.63319, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 2
binning_y: 2
roi: 
  x_offset: 0
  y_offset: 0
  height: 2050
  width: 2448
  do_rectify: False
```

Specifically matrices K and P should have their non-homogeneous components scaled by the decimation, the height and width should also be scaled.

vrabaud, 2015-03-22:
well, not according to the doc: http://docs.ros.org/indigo/api/sensor_msgs/html/msg/CameraInfo.html
binning and roi are just operational parameters.

cbames, 2015-03-24:
I see, my mistake, If I grab the K matrices using  PinholeCameraModel::fromCameraInfo from image_geometry then it accounts for the binning.  Thanks, and sorry for inconvenience.

closed 2015-03-24 by cbames

fabrizioschiano, 2016-11-02:
Hi, 
I know this is an old post. But how did you solve your problem? I know it is supposed to work like that but how I can inform the receiver of my camera_dec/image_raw message that it is receiving an image which has been decimated? 

Thanks in advance.