ros-perception/image_pipeline #123: Stereo matching with semi-global block matching algorithm

davidfornas, 2015-04-16:
I don't know if it is possible to do it and how slow will it be. However, having a precise point cloud when a one off point cloud is needed could be very useful. I'm eager to implement it myself, but I'm wondering if it could be included here or then it is a separate package.

vrabaud, 2015-04-16:
don't implement it, use the one in OpenCV: http://docs.opencv.org/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html#stereosgbm
Just add an option in the cfg file to also include that method. Or maybe a different launch file that would call a different config file

davidfornas, 2015-04-16:
Yes, I was referring to that, switch StereoBM for StereoSGBM in https://github.com/ros-perception/image_pipeline/blob/indigo/stereo_image_proc/include/stereo_image_proc/processor.h and then make the necessary changes in the rest of the code. I'll try to make it in a fork and then you can merge it if you wish.

davidfornas, 2015-05-15:
I have seen that this has been addressed here https://github.com/ros-perception/image_pipeline/commit/067e86ab8d77f87533dbb6c09a219df7b452d981 Why this option is not used in the deb packages nor included in Indigo? This implementation skips the processor.cpp usage but it works, although a bit slow.

vrabaud, 2015-05-16:
well, I simply did not do my job as a maintainer :)
That commit was the only one that was not forward-ported. Just did in 79ed0664e8cf1e3c1d7948c65ee0a9be405e0d38
Can you please let me know if that works for you ? Thx

kanirudh54, 2015-06-03:
Can this change also be made in Hydro branch?

njzhangyifei, 2015-06-04:
I believe these has been implemented in the hydro-devel branch. https://github.com/ros-perception/image_pipeline/commit/6c343e9416ea6e5e22b44e6450f1b11b79a571b0

kanirudh54, 2015-06-05:
ok. But how do I use dynamic reconfigure to use stereoSGBM. I tried rosrun dynamic_reconfigure dynparam get /stereo/stereo_processing and got 

{'disparity_range': 128, 'speckle_size': 349, 'groups': {'disparity_range': 128, 'speckle_size': 349, 'correlation_window_size': 13, 'parent': 0, 'prefilter_cap': 31, 'texture_threshold': 10, 'groups': {}, 'uniqueness_ratio': 15.0, 'id': 0, 'name': 'Default', 'parameters': {}, 'speckle_range': 4, 'min_disparity': -128, 'state': True, 'prefilter_size': 9, 'type': ''}, 'prefilter_size': 9, 'speckle_range': 4, 'min_disparity': -128, 'texture_threshold': 10, 'correlation_window_size': 13, 'prefilter_cap': 31, 'uniqueness_ratio': 15.0}

Is that the right node? I also tried rqt_reconfigure but didn't find any parameter which can change from stereoBM to stereoSGBM.

njzhangyifei, 2015-06-05:
To use the hydro-devel branch of the image_pipeline, you will have to eventually get a clone (or a submodule) of the image_pipeline repo in your ros workspace. Then you will need to switch it to hydro-devel branch and do a catkin_make.

kanirudh54, 2015-06-05:
ok, I installed it via sudo apt-get install ros-hydro-stereo-image-proc . So this way doesn't work?

kanirudh54, 2015-06-05:
I think it does't work that way(apt-get install). I had done as you had said and it worked. Now I get new parameters. Thanks

malban, 2015-07-12:
Does the version number of the stereo_image_proc package need to get incremented on the indigo branch to trigger a new debian being built?

vrabaud, 2015-08-02:
1.12.14 got released for Indigo.

closed 2015-08-02 by vrabaud