ros-perception/image_pipeline #124: When having opencv3 installed stereo calibration fails

awesomebytes, 2015-04-17:
This is not really an issue as the error is on the user side but it may help someone else that finds the same error.

Having `ros-indigo-opencv3` installed when I tried to use the calibration package for a pair of stereo cameras I got the error:

    SystemError: new style getargs format but argument is not a tuple

Which is quite cryptic. I tried to fix it playing with the parameters to give to cv2.stereoCalibration but I couldn't... until I realized that I had installed opencv3 which was first in the path, so when I did 

    import cv2
    cv2.__version__

Version 3.0 appeared. I needed to uninstall `ros-indigo-opencv3` in order to make it work.

I hope this Issue at least appears in some google search related to this (so the next person does not lose so much time as I did).

vrabaud, 2015-04-17:
do you have an exact line where it appears ?

awesomebytes, 2015-04-17:
Exactly here:
https://github.com/ros-perception/image_pipeline/blob/indigo/camera_calibration/src/camera_calibration/calibrator.py#L849-L855

w4zir, 2016-01-17:
how did you resolve it ?

vrabaud, 2016-01-17:
This actually been fixed in e1bd3a058a9d566718475f7557b98050bf304060

closed 2016-01-17 by vrabaud