ros-perception/image_pipeline #125: QMetaMethod::invoke: Unable to invoke methods with return values in queued connections

alexus37, 2015-04-24:
Hi,
I have a problem using image_view on my system (Trusty, indigo). 
When I try to get a image by running

rosrun image_view image_view image:=/IR_data

everytime a new frame arives I get this error:

QMetaMethod::invoke: Unable to invoke methods with return values in queued connections

Do you have any idea want could cause this?!

Thanks for your help.

Btw I have opencv 2.4.9 installed

vrabaud, 2015-04-25:
did you compile OpenCV yourself ? Which OS ?

alexus37, 2015-04-26:
Yes I compiled opencv myself following this guide 

http://www.samontab.com/web/2014/06/installing-opencv-2-4-9-in-ubuntu-14-04-lts/

My OS is Ubuntu 14.04

alexus37, 2015-04-30:
hi,
I found a solution, I just recompiled opencv 2.4.9 without QT!
That fix it for me!

closed 2015-05-16 by vrabaud