ros-perception/image_pipeline #126: why stereo_camera_calibration epi result is "?"

LEE-JAE-SEONG, 2015-04-28:
Hi, 
my system are ubuntu 14.04, 64bit, Indigo
after stereo camera calibrate, epi value is "?" 
why is this result ?

vrabaud, 2015-05-11:
what do you mean by epi ? Where do you see that result ?

theDrGray, 2015-10-05:
Was there ever a solution to this issue? 

I get the same result when calibrating stereo monochrome Flea3 cameras on Ubuntu 14.04.1 64 on Indigo. 

I run "camera_calibration cameracalibrator.py" with the proper parameters for stereo. Everything appears to go great. I am able to get tons of samples (170+) pretty quickly (< 5 min). From there I wait roughly 30 minutes for the calibration to succeed. It spits out usable .yaml files and valid projections, but epi value above the "Calibrate" button to the right of the UI is a question mark. I also have not seen a disparity value at all. 

When I launch viso2_ros after using those yaml files to launch the driver viso2 is able to subscribe to all the topics it needs, but I get the low number of synced tuples error. Not sure if these issues are related at all, but semi hoping they are.

trainman419, 2015-10-05:
After running the camera calibration, it will only calculate the epi value if there is a checkerboard in the camera's field of view; otherwise it just shows "?" if if can't detect the checkerboard.

JWhitleyAStuff, 2019-04-25:
Closed due to resolution.

closed 2019-04-25 by JWhitleyAStuff