ros-perception/image_pipeline #127: Is that possible to view BGRA/RGBA image with image_view?

wkentaro, 2015-05-10:
I'd like to view image which encoding is BGRA or RGBA with image_view.
Is that possible? Or do I need to change some codes?

vrabaud, 2015-05-10:
that would be a question for http://answers.ros.org/questions/ . But it does yes. Does it not work for you ?

wkentaro, 2015-05-11:
> that would be a question for http://answers.ros.org/questions/ .

Sorry about that.

It can be viewed but the transparent pixels does not viewed as transparent.

Left is image_view, Right is png image viewed with Image Viewer on Ubuntu.
[image]

vrabaud, 2015-05-11:
ok, so your question if more of a bug report :)
The transparent pixels are actually converted to black (that is the OpenCV convention). What would you expect ?

wkentaro, 2015-05-12:
I see. This may be confusing but I'd expect the transparent pixels are
converted to the same color as image_view's bg (I don't know the color code
but it's gray.)

wkentaro, 2018-05-16:
I'm closing this since I don't need this feature anymore. (actually, I forgot why I needed this..)
Thanks for the comment!

closed 2018-05-16 by wkentaro