ros-perception/image_pipeline #128: Unable to launch calibration window

chrisspen, 2015-05-11:
I'm running Indigo on Ubuntu 14.04, and I attempted to follow the [stereo calibration tutorial](http://wiki.ros.org/camera_calibration/Tutorials/StereoCalibration), but I could not reliably get the calibration window to launch despite receiving no errors.

I'm using two USB webcams configured with the launch file:

    <launch>
        <arg name="device_left" default="/dev/video1" />
        <arg name="device_right" default="/dev/video2" />
        <arg name="width" default="640" />
        <arg name="height" default="480" />
        <arg name="format" default="yuyv" />
        <group ns="raw_stereo">
            <node name="left" pkg="usb_cam" type="usb_cam_node" output="screen">
                <param name="video_device" value="$(arg device_left)" />
                <param name="image_width" value="$(arg width)" />
                <param name="image_height" value="$(arg height)" />
                <param name="pixel_format" value="$(arg format)" />
                <param name="camera_frame_id" value="left" />
                <param name="io_method" value="mmap" />
            </node>
            <node name="right" pkg="usb_cam" type="usb_cam_node" output="screen">
                <param name="video_device" value="$(arg device_right)" />
                <param name="image_width" value="$(arg width)" />
                <param name="image_height" value="$(arg height)" />
                <param name="pixel_format" value="$(arg format)" />
                <param name="camera_frame_id" value="right" />
                <param name="io_method" value="mmap" />
            </node>
        </group>
    </launch>

I then tried running the calibrator with:

    rosrun camera_calibration cameracalibrator.py --size 8x6 --square 0.025 right:=/raw_stereo/right/image_raw left:=/raw_stereo/left/image_raw right_camera:=/raw_stereo/right left_camera:=/raw_stereo/left

The terminal reports:

    ('Waiting for service', '/raw_stereo/left/set_camera_info', '...')
    OK
    ('Waiting for service', '/raw_stereo/right/set_camera_info', '...')
    OK

The calibration GUI appeared initially, but was very slow, appeared to freeze and then crashed with a segfault. Re-running the command now results in no GUI ever appearing.

Since I'm using unsynchronized cameras, I also tried running:

    rosrun camera_calibration cameracalibrator.py --size 8x6 --square 0.025 right:=/raw_stereo/right/image_raw left:=/raw_stereo/left/image_raw right_camera:=/raw_stereo/right left_camera:=/raw_stereo/left  --approximate=0.01

but the result is the same.

chrisspen, 2015-05-11:
The problem was a combination of a large resolution (causing the slowness) coupled with not using a large enough `--approximate` value.

closed 2015-05-11 by chrisspen

oaix, 2018-02-28:
‘Waiting for service‘: add the setting "--no-service-check", that is:
rosrun camera_calibration cameracalibrator.py --size 8x6 --square 0.025 right:=/raw_stereo/right/image_raw left:=/raw_stereo/left/image_raw right_camera:=/raw_stereo/right left_camera:=/raw_stereo/left --no-service-check