ros-perception/image_pipeline #130: Stereo_image_proc does not support unsynchronized cameras

chrisspen, 2015-05-11:
I'm able to calibrate a stereo camera by using `cameracalibrator.py` with an `--approximate=0.1`. However, if I attempt to run a stereo_image_proc node with `approximate_sync:=true, it processes no images even though image_view confirms the calibrated cameras are publishing images. My guess is that stereo_image_proc doesn't support the same approximate threshold as `cameracalibrator.py` and whatever threshold it's using is too low, causing no stereo images to be processed.

My launch file to start the webcams:

    <launch>
        <arg name="device_left" default="/dev/video1" />
        <arg name="device_right" default="/dev/video2" />
        <arg name="width" default="320" />
        <arg name="height" default="240" />
        <arg name="format" default="yuyv" />
        <group ns="raw_stereo">
            <node name="left" pkg="usb_cam" type="usb_cam_node" output="screen">
                <param name="video_device" value="$(arg device_left)" />
                <param name="image_width" value="$(arg width)" />
                <param name="image_height" value="$(arg height)" />
                <param name="pixel_format" value="$(arg format)" />
                <param name="camera_frame_id" value="left_camera" />
                <param name="io_method" value="mmap" />
                <param name="camera_info_url" type="string" value="file://$(find stereo_slam_test)/config/usb1-config-left.yml" />
            </node>
            <node name="right" pkg="usb_cam" type="usb_cam_node" output="screen">
                <param name="video_device" value="$(arg device_right)" />
                <param name="image_width" value="$(arg width)" />
                <param name="image_height" value="$(arg height)" />
                <param name="pixel_format" value="$(arg format)" />
                <param name="camera_frame_id" value="right_camera" />
                <param name="io_method" value="mmap" />
                <param name="camera_info_url" type="string" value="file://$(find stereo_slam_test)/config/usb1-config-right.yml" />
            </node>
        </group>
    </launch>

The command I use to start stereo_image_proc:

    ROS_NAMESPACE=raw_stereo rosrun stereo_image_proc stereo_image_proc approximateync:=true

The command I use to view the output from stereo_image_proc:

    rosrun image_view stereo_view stereo:=/raw_stereo image:=image_rect_color

This outputs a large number of warnings and errors like:

    [ WARN] [1431314330.738338807]: [stereo_view] Low number of synchronized left/right/disparity triplets received.
    Left images received:      3120 (topic '/raw_stereo/left/image_rect_color')
    Right images received:     3121 (topic '/raw_stereo/right/image_rect_color')
    Disparity images received: 0 (topic '/raw_stereo/disparity')
    Synchronized triplets: 0
    Possible issues:
	    * stereo_image_proc is not running.
	      Does `rosnode info /stereo_view_1431314030409464242` show any connections?
	    * The cameras are not synchronized.
	      Try restarting stereo_view with parameter _approximate_sync:=True
	    * The network is too slow. One or more images are dropped from each triplet.
	      Try restarting stereo_view, increasing parameter 'queue_size' (currently 5)

BenBlumer, 2015-09-23:
I have the same issue with the latest release running Igloo.  I can view the rectified images, but I can't view the disparity image and stereo view reports left & right images received, but no disparity images.

BenBlumer, 2015-09-23:
I have a similar issue: running stereo_image_proc without the approximatesync parameter, stereoview gets the left and right images, but no disparity images (although it also won't display the left and right images).  Running stereo_image_proc with the approximatesync parameter results in stereo_image_proc crashing when I `rosrun image_view stereo_view stereo:=/raw_stereo image:=image_rect_color`.

 The crash is `vision_opencv/image_geometry/src/stereo_camera_model.cpp:35: bool image_geometry::StereoCameraModel::fromCameraInfo(const CameraInfo&, const CameraInfo&): Assertion 'left_.tfFrame() == right_.tfFrame()' failed.
Aborted (core dumped)`

sophialiu416, 2016-11-29:
I have the exact same problem now, I tried different approximate, but still doesn't work. The rosnode info /stereo_view_1431314030409464242 shows 
contacting node http://tesla:38974/ ...
Pid: 7258
Connections:
 * topic: /rosout
    * to: /rosout
    * direction: outbound
    * transport: TCPROS
 * topic: /stereo/left/image_rect
    * to: /stereo/stereo_image_proc (http://tesla:44114/)
    * direction: inbound
    * transport: TCPROS
 * topic: /stereo/right/image_rect
    * to: /stereo/stereo_image_proc (http://tesla:44114/)
    * direction: inbound
    * transport: TCPROS
 * topic: /stereo/disparity
    * to: /stereo/stereo_image_proc (http://tesla:44114/)
    * direction: inbound
    * transport: TCPROS
Anybody can solve the problem?

k-okada, 2016-12-03:
Try
```
ROS_NAMESPACE=raw_stereo rosrun stereo_image_proc stereo_image_proc _approximate_sync:=true
rosrun image_view stereo_view stereo:=/raw_stereomage:=image_rect_color _approximate_sync:=true
```
If this works, where did you find `approximateync:=true` option?

evorta, 2017-02-19:
Hi all,

A similar issue, 2 webcams, Linux 16.04, Ros Kinetic.

Launched:
`roslaunch low_res_stereo_cam.launch
`
It correctly gives the left/image_raw, etc along with camera_info. All good there, can see the video in:
```
rosrun image_view image_view image:=/stereo/right/image_raw
rosrun image_view image_view image:=/stereo/left/image_raw
```

Then i started the stereo_image_proc
`ROS_NAMESPACE=stereo rosrun stereo_image_proc stereo_image_proc _approximate_sync=true`

I can see the correct greyscale video on:
rosrun image_view image_view image:=/stereo/right/image_rect
rosrun image_view image_view image:=/stereo/left/image_rect

However disparity seems empty. Running:

`rosrun image_view stereo_view stereo:=stereo image:=image_rect _approximate_sync:=true`

shows frames from left and right being picked up but nada on the disparity front. Any ideas ? 

Here is an example of the debug from the above while running, the disparity images received is always 0.

```

[ WARN] [1487477520.200687541]: [stereo_view] Low number of synchronized left/right/disparity triplets received.
Left images received:      7873 (topic '/stereo/left/image_rect')
Right images received:     6276 (topic '/stereo/right/image_rect')
Disparity images received: 0 (topic '/stereo/disparity')
Synchronized triplets: 0
Possible issues:
	* stereo_image_proc is not running.
	  Does `rosnode info /stereo_view_1487477234995986346` show any connections?
	* The cameras are not synchronized.
	  Try restarting stereo_view with parameter _approximate_sync:=True
	* The network is too slow. One or more images are dropped from each triplet.
	  Try restarting stereo_view, increasing parameter 'queue_size' (currently 5)
[ WARN] [1487477535.200803855]: [stereo_view] Low number of synchronized left/right/disparity triplets received.
Left images received:      8323 (topic '/stereo/left/image_rect')
Right images received:     6650 (topic '/stereo/right/image_rect')
Disparity images received: 0 (topic '/stereo/disparity')
Synchronized triplets: 0
Possible issues:
	* stereo_image_proc is not running.
	  Does `rosnode info /stereo_view_1487477234995986346` show any connections?
	* The cameras are not synchronized.
	  Try restarting stereo_view with parameter _approximate_sync:=True
	* The network is too slow. One or more images are dropped from each triplet.
	  Try restarting stereo_view, increasing parameter 'queue_size' (currently 5)
```

evorta, 2017-02-19:
Hi all, I solved this and have it all working... if it helps I have pasted my full set of configuration. The remapping isn't probably too wise but it gets it working :) please do correct me if you know of a better way?

```

<launch>
<group ns="stereo">

  <node name="left" pkg="usb_cam" type="usb_cam_node" output="screen" >
    <param name="video_device" value="/dev/video0" />
    <param name="image_width" value="640" />
    <param name="image_height" value="480" />
    <param name="pixel_format" value="yuyv" />
    <param name="camera_frame_id" value="left" />
    <param name="io_method" value="mmap"/>
    <param name="camera_info_url" value="file:///home/MYHOME/Documents/left.yml"/>
  </node>

  <node name="right" pkg="usb_cam" type="usb_cam_node" output="screen" >
    <param name="video_device" value="/dev/video1" />
    <param name="image_width" value="640" />
    <param name="image_height" value="480" />
    <param name="pixel_format" value="yuyv" />
    <param name="camera_frame_id" value="right" />
    <param name="io_method" value="mmap"/>
    <param name="camera_info_url" value="file:///home/MYHOME/Documents/right.yml"/>
  </node>

<node name="stereo_image_proc" pkg="stereo_image_proc" type="stereo_image_proc" output="screen"  >
<param name="queue_size" value="2000" />
<param name="approximate_sync" value="True" />
<remap from="/stereo/left/image" to="/stereo/left/image_raw"/>
<remap from="/stereo/right/image" to="/stereo/right/image_raw"/>
</node>

</group>

<node name="stereo_cam" pkg="image_view" type="stereo_view" output="screen" >
<param name="queue_size" value="2000" />
<param name="approximate_sync" value="True" />
<remap from="/stereo/left/image" to="/stereo/left/image_raw"/>
<remap from="/stereo/right/image" to="/stereo/right/image_raw"/>
</node>

</launch>
```

JWhitleyAStuff, 2019-04-25:
Closed due to resolution.

closed 2019-04-25 by JWhitleyAStuff