ros-perception/image_pipeline #133: Cmake Error on ros-hydro on Ubuntu 12.04, can't seem to identify if I'm missing dependencies maybe ?

Assadkhanjaffer, 2015-05-15:
root@assad-Studio-1558:~/catkin_ws# catkin_make
Base path: /home/assad/catkin_ws
Source space: /home/assad/catkin_ws/src
Build space: /home/assad/catkin_ws/build
Devel space: /home/assad/catkin_ws/devel
Install space: /home/assad/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/assad/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/assad/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/assad/catkin_ws/devel;/opt/ros/hydro
-- This workspace overlays: /home/assad/catkin_ws/devel;/opt/ros/hydro
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Python version: 2.7
-- Using Debian Python package layout
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/assad/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- catkin 0.5.90
-- BUILD_SHARED_LIBS is on
WARNING: Package "libg2o" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 13 packages in topological order:
-- ~~  - camera_umd (metapackage)
-- ~~  - image_pipeline (metapackage)
-- ~~  - tld_msgs
-- ~~  - camera_calibration
-- ~~  - image_view
-- ~~  - jpeg_streamer
-- ~~  - tld_tracker
-- ~~  - stereo_image_proc
-- ~~  - depth_image_proc
-- ~~  - image_rotate
-- ~~  - tum_ardrone
-- ~~  - usb_cam
-- ~~  - uvc_camera
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'camera_umd'
-- ==> add_subdirectory(camera_umd/camera_umd)
-- +++ processing catkin metapackage: 'image_pipeline'
-- ==> add_subdirectory(image_pipeline/image_pipeline)
-- +++ processing catkin package: 'tld_msgs'
-- ==> add_subdirectory(ros_opentld/tld_msgs)
-- Using these message generators: gencpp;genlisp;genpy
-- tld_msgs: 2 messages, 0 services
-- +++ processing catkin package: 'camera_calibration'
-- ==> add_subdirectory(image_pipeline/camera_calibration)
-- +++ processing catkin package: 'image_view'
-- ==> add_subdirectory(image_pipeline/image_view)
-- +++ processing catkin package: 'jpeg_streamer'
-- ==> add_subdirectory(camera_umd/jpeg_streamer)
-- +++ processing catkin package: 'tld_tracker'
-- ==> add_subdirectory(ros_opentld/tld_tracker)
-- +++ processing catkin package: 'stereo_image_proc'
-- ==> add_subdirectory(image_pipeline/stereo_image_proc)
CMake Error at /home/assad/catkin_ws/devel/share/image_proc/cmake/image_procConfig.cmake:106 (message):
  Project 'image_proc' specifies
  '/home/assad/catkin_ws/src/image_pipeline/image_proc/include' as an include
  dir, which is not found.  It does neither exist as an absolute directory
  nor in
  '/home/assad/catkin_ws/src/image_pipeline/image_proc//home/assad/catkin_ws/src/image_pipeline/image_proc/include'.
  Ask the maintainer 'Vincent Rabaud <vincent.rabaud@gmail.com>' to fix it.
Call Stack (most recent call first):
  /opt/ros/hydro/share/catkin/cmake/catkinConfig.cmake:75 (find_package)
  image_pipeline/stereo_image_proc/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

vrabaud, 2015-05-15:
Well, I believe you did something funky with your workspace: your CMake error is referencing ```image_proc``` but it is not part of the 13 packages you are building.
Just delete your build and devel folder and recompile everything. That should fix it.

vrabaud, 2015-06-20:
please re-open if it still happens.

closed 2015-06-20 by vrabaud