ros-perception/image_pipeline #134: cameracheck.py error when checkerboard gets in focus.

bjornph, 2015-06-03:
Running ubuntu 14.04 and ros  indigo.

I  run
 rosrun camera_calibration cameracheck.py --size 11x7 --square 0.04 monocular:=/usb_cam
it prints no  chessboard continiusly when there's no chessboard in my image  frame. When it does get it fails  with:

Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/opt/ros/indigo/lib/camera_calibration/cameracheck.py", line 83, in run
    self.function(m)
  File "/opt/ros/indigo/lib/camera_calibration/cameracheck.py", line 156, in handle_monocular
    if C:
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()

What's wrong?

njzhangyifei, 2015-06-08:
This issue should have been fixed by https://github.com/ros-perception/image_pipeline/commit/75ecbaa018017a3868cbe0922ad3f04aa4338b1d

vrabaud, 2015-07-25:
indeed and that just got released in shadow-fixed. Fixed in 1.12.14

closed 2015-07-25 by vrabaud