ros-perception/image_pipeline #135: After commit the calibration data to camera all driver segment fault

StanleyYake, 2015-06-09:
Hello there:
    when I calibration the camera and push the button commit(really regret), then I convert the ost.txt to camera.yaml and put it in the ./ros/camera_info directory and `rosrun uvc_cam uvc_cam_node` throw Segment Fault.And when I remove the camera.yaml it throw the same error.also I have tried some other drivers ,nothing good happen, but when I use another computer, and didn't calibrate, it could open the camera.I wonder if there is another config file in my computer and all the driver will load in advance, but the config is wrong or not so good ,so all driver can't open the camera.By the way, I could use the app cheese to open it ,but the color is a little strange.
[image]
[image]

StanleyYake, 2015-06-09:
[image]

vrabaud, 2015-06-27:
do you have the obtained calibration file ? Do you have the default one from uvc_cam ?

StanleyYake, 2015-06-28:
Hi vrabaud:
    Thanks for your answer. After calibration, I get a compressed file, and there is a file named ost.txt in it which is the calibration file.But I didn't have the default one , I thought it should be in the camera firmware.
    BTW, I found another package `libuvc_camera` http://wiki.ros.org/libuvc_camera which is the only one work on my computer

JWhitleyAStuff, 2019-04-25:
Closed due to resolution.

closed 2019-04-25 by JWhitleyAStuff