ros-perception/image_pipeline #136: Image View on Ubuntu 14.04 using ROS Indigo

kanirudh54, 2015-06-11:
When I run the  **rosrun image_view image_view image:=/duo3d_camera/left/image_raw**
I get the following error:
[ INFO] [1433967448.442631200]: Using transport "raw"
init done 
opengl support available 

(image_view:31056): GLib-GObject-WARNING **: invalid uninstantiatable type '(null)' in cast to 'GtkWidget'

(image_view:31056): GLib-GObject-WARNING **: instance of invalid non-instantiatable type '(null)'

(image_view:31056): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

I had followed the other [issue](https://github.com/ros-perception/image_pipeline/issues/114) an tried 1.12.10 version which was working, but had problems with other package I wanted to use

vrabaud, 2015-06-11:
You are using your own OpenCV right, not the Ubuntu one ?

kanirudh54, 2015-06-11:
I have also installed OpenCV 2.4.11 on my system. But I also have the Ubuntu packages of OpenCV installed via apt-get. Should I remove the OpenCV?

vrabaud, 2015-06-11:
this error is due to having compiled OpenCV in a different way from what Ubuntu does. Therefore, your configuration has not been tested and that creates that error. If you remove your 2.4.11, it should run properly. Any reason for using 2.4.11 ? If you need the latest and greatest, you could also link to the OpenCV 3 packages that were released in ROS.

kanirudh54, 2015-06-11:
The only reason for using OpenCV was to use the [Duo3d](https://duo3d.com/product/duo-mini-lv1#tab=docs) SDK

vrabaud, 2015-06-11:
and it requires 2.4.11 ? What features does it need that are not in 2.4.8 ?

kanirudh54, 2015-06-11:
It didn't specifically require 2.4.11. The [Duo3d SDK](https://duo3d.com/docs/articles/sdk) required 2.4.7+, so I separately compiled it. But [later](https://github.com/l0g1x/DUO-Camera-ROS/issues/18) I realized that the Ubuntu one was good enough. It presently works as expected after making changes.

vrabaud, 2015-06-14:
thx, closing then.

closed 2015-06-14 by vrabaud