ros-perception/image_pipeline #139: How to  downgrade the image_pipeline

5aou, 2015-06-22:
Hi all, 
I'm trying to install ORB_SLAM on ubuntu 14.04 + Ros Indigo but I  encounter this issue : 
(image_view:23374): GLib-GObject-WARNING **: invalid uninstantiatable type '(null)' in cast to 'GtkWidget'

(image_view:23374): GLib-GObject-WARNING **: instance of invalid non-instantiatable type '(null)'

(image_view:23374): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

I searched in the Internet and I found that I've to downgrade the image_pipeline but I don't know how to do it.

Thank you a lot.

vrabaud, 2015-06-27:
Have you compiled your own OpenCV ? (maybe it is in /use/loca/lib)

5aou, 2015-06-28:
Thank you a lot for your response , I solved that error, now  I encounter a segfault one, when I do : rosrun ORB_SLAM ORB_SLAM............... Any suggestions please. 

Thank you again.

vrabaud, 2015-07-02:
orb_slam is not part of image_pipeline so you should report bugs to the creators of that package.

closed 2015-07-02 by vrabaud