ros-perception/image_pipeline #140: Stereo calibration problem

hbadino, 2015-06-24:
Hi,

I've been dealing with this issue for a while now and I can't think of any particular reason why calibration  fails. The optimization fails, it finds a incorrect set of camera parameters leading to the rectified epipoles projected within the image FOV  as opposed to lateral infinity as they should in a standard stereo configuration. This happens every single time, no matter how many times I tried to recalibrate, no matter how many parameters I want to optimize.

My setup and some observations:

 - Cameras are physically well aligned in a left/right configuration. 
 - Chessboard detector (7x5 with square size 7.34 cm) seems to work fine (with the exception of some possible issue that I mention to below).
 - Display bars for X, Y, and Skew go green with full range. Size goes green at 2/3 range.
 - more than 150 frames are selected before optimization starts (I've tried less and the results is the same) 
 - I've been using -k 3, but it also fails on -k 5 and -k 1
 - The found Intrinsics and extrinsic parameters seems to be reasonable, with the exception of the depth component of the translation which is extremely large (in the range [0.15,0.2]). The real physical distance is around 1 mm.

The only weird thing I see from the interface is that the detector sometimes finds the chessboard, but the color configuration is switched between left and right. Very often, on the left image, the detector finds the chessboard with colors red,orange,yellow,green and cyan from top to bottom, while in the right image the order is reversed. I wonder if this is the cause of the problem. If the color is an indication of the corresponding relative order of the detected chessboard points, it would have wrong data point associations and can lead to many outliers leading to optimization divergence.

Any help would be greatly appreciated.

Hernan

vrabaud, 2015-06-27:
If there is indeed a flip, you should probably use an asymmetric grid .
Do you have a rosbag with this data ? I do not have a stereo camera.

hbadino, 2015-06-29:
Vincent,

thanks for helping. What do you mean with asymmetric grid?

I do have a bag file but it is 8 GB. Is there anywhere I can upload it? 

I can provide you with the individual files provided in the output, but I'm not sure if you will be able to replicate the problem that way.

Hernan

vrabaud, 2015-07-16:
for the bag, just compress the images or split it using the tutorials. Just put it on a Google drive and share it with me.

cwecht, 2016-12-20:
I had the same problem. I was able to fix this by forcing the first corner of the detected chessboard to be in the upper left. I've done this [here](https://github.com/cwecht/image_pipeline/commit/887e1b3966128e4e4eef37abae4da075df39fa35) and it worked pretty well for me.

catta202000, 2016-12-20:
I had the same problem, i solved forcing the first corner to be the upper left too.

```python
if(corners[0][0][0] > corners[len(corners)-1][0][0]):`
    corners = corners[::-1,:]`
    corners = corners.copy()
```

vrabaud, 2017-01-02:
@cwecht , @catta202000 , don't hesitate to make a pull request when you have a solution ! Thx.

commit f94bfe7b (2017-01-02): Force first corner off chessboard to be uppler left.
Fixes #140
closed 2017-01-02 by vrabaud