ros-perception/image_pipeline #143: rotate node could have a simple option of flipping image

awesomebytes, 2015-07-20:
Being able to just run it to flip horizontall and/or vertically an image sounds like a common use case to me.

Maybe an special flag could be set for this. Creating a TF frame just for that may be a bit of an overkill.

vrabaud, 2015-07-20:
well, either we flip the tf, either we flip the image :) I agree with flipping the image, that seems valuable.

awesomebytes, 2015-07-20:
I made an implementation at #145, you can check the [README](https://github.com/awesomebytes/image_pipeline/blob/add_image_flip/image_flip/README.md), it's a standalone node used as:

    rosrun image_flip image_flip image:=<image topic> flipped:=<flipped image topic> <h(orizontal)/v(ertical)/b(oth)>

And not flags of rotate as I think rotate has a specific use and adding flags for this would be weird.

vrabaud, 2016-07-24:
fixed in #173

closed 2016-07-24 by vrabaud