ros-perception/image_pipeline #146: support fisheye calibration using opencv 3.0

v01d, 2015-08-17:
OpenCV 3.0 now has support for fisheye lens calibration and related distortion/projection functions.
This could be used to extend ROS's support towards calibrating and processing images acquired with fisheye lenses.

http://docs.opencv.org/3.0-beta/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html#fisheye

I guess one possibility is to add another distortion model keyword, which would indicate these additional functions need to be used. I believe only a skew coefficient is needed as a separate parameter for calibration.

DavidTorresOcana, 2018-08-14:
Hello
Perhaps too late, but I added this functionality in this fork: [image_pipeline](https://github.com/DavidTorresOcana/image_pipeline)

You can calibrate (with a tool) and rectify fisheye cameras.

The results for high FOV (+180deg) are better that those achieved with the current implementation

DaikiMaekawa, 2018-10-24:
@DavidTorresOcana Thanks for sharing. I think fisheye is one of the things that has to be supported by image_pipeline. So, it would be nice to include your changes into the upstream repo. What's stands in the way of creating a PR?

Roemer, 2019-10-23:
Any news here? That feature could be really handy.

SteveMacenski, 2019-10-23:
@roemer PRs welcome

PR #440 (2019-11-07): Add Fisheye calibration tool
This PR introduces changes to be able to calibrate fisheye cameras and aim to Fix #146

This PR is related with ros/common_msgs#151 , ros-perception/vision_opencv#306 and ros-perception/image_common#146

Main changes it introduces are:
 * Upgrade current calibration tool to:
     - Allow user select between pinhole and fisheye model
     - Calibrate Monocular and Stereo with fisheye model
commit 1cdc1284 (2019-11-07): Add Fisheye calibration tool (#440)

* Add Fisheye calibration tool

* Add calibration for fisheye cameras
Fix #146

* Correct typo

* Restore camera_calib files permisions

* Upgrades to calibrator tool for multi model calibration

* Solve fisheye balance selection TODO:
For some reason estimateNewCameraMatrixForUndistortRectify is not producing the expected result, hence a workaround was implemented

* Add fisheye calibration flags as user arguments

* Add undistortion of points for fisheye

* cam_calib: rolling back flags
Rolling back changes to previous commit on camera calibrator flags to [... cut]
commit 5f62e1be (2021-11-11): Add Fisheye calibration tool (#440)

* Add Fisheye calibration tool

* Add calibration for fisheye cameras
Fix #146

* Correct typo

* Restore camera_calib files permisions

* Upgrades to calibrator tool for multi model calibration

* Solve fisheye balance selection TODO:
For some reason estimateNewCameraMatrixForUndistortRectify is not producing the expected result, hence a workaround was implemented

* Add fisheye calibration flags as user arguments

* Add undistortion of points for fisheye

* cam_calib: rolling back flags
Rolling back changes to previous commit on camera calibrator flags to [... cut]
closed 2019-11-07 by JWhitleyAStuff