ros-perception/image_pipeline #148: QMetaMethod::invoke: Unable to invoke methods with return values in queued connections

TSC21, 2015-08-29:
Hi there,

I'm launching a usb camera node with has the following in the launch file:
```xml
<launch>
  <node name="usb_cam" pkg="usb_cam" type="usb_cam_node" output="screen" >
    <param name="video_device" value="/dev/video0" />
    <param name="image_width" value="640" />
    <param name="image_height" value="480" />
    <param name="pixel_format" value="mjpeg" />
    <param name="camera_frame_id" value="usb_cam" />
    <param name="io_method" value="mmap"/>
  </node>
  <node name="image_view" pkg="image_view" type="image_view" respawn="false" output="screen">
    <param name="autosize" value="true" />
  </node>
</launch>
```
On the terminal where I launch it, when it runs the `image_view` node I get:
```bash
QMetaMethod::invoke: Unable to invoke methods with return values in queued connections
```
Supposedly this was something corrected on a previous release but I'm getting it with version 1.12.14. I'm on ROS Indigo, running on Ubuntu 14.04 LTS.

Thanks in advance!

vrabaud, 2015-08-31:
do you have your own compiled OpenCV ?

TSC21, 2015-08-31:
No. I'm using ROS one.

vrabaud, 2015-09-01:
I totally missed you are using OpenCV3: image_pipeline is compiled with OpenCV2. Therefore, if you install the OpenCV3 package (which has a totally different API), it will try to use it and crash. I should probably have the packages conflict.

closed 2015-09-01 by vrabaud