ros-perception/image_pipeline #149: Give priority to custom build OpenCV

aleksandaratanasov, 2015-09-17:
Hi!

I'm currently struggling to get OpenCV work with ROS Indigo on Debian Jessie. I have noticed the following - almost (note the "almost" part) all CMakeLists.txt give priority to the libraries build with catkin. I've noticed that in some however the includes and libraries for OpenCV come before those of catkin. I find this inconsistent and also making things difficult to use custom builds of OpenCV when you want to update the image_pipeline to use your own installation of OpenCV. In order to change that I suggest that everywhere where OpenCV's includes and libraries are called that we give those a priority over catkin (this might not be restricted only to the image_pipeline packages).

Example:

    include_directories(SYSTEM ${catkin_INCLUDE_DIRS} ${OpenCV_INCLUDE_DIRS})

should become

    include_directories(SYSTEM ${OpenCV_INCLUDE_DIRS} ${catkin_INCLUDE_DIRS})

and

    target_link_libraries(${PROJECT_NAME} ${catkin_LIBRARIES} ${Boost_LIBRARIES} ${OpenCV_LIBRARIES})

should become

    target_link_libraries(${PROJECT_NAME} ${Boost_LIBRARIES} ${OpenCV_LIBRARIES} ${catkin_LIBRARIES})

This will give custom build OpenCV a priority when building the selected package. I am no cmake expert so I'd like to discuss this here whether what I'm saying is true. I'm currently also trying to get this to work with PCL (not post-installation but building the ROS PCL packages from source with a custom version of PCL). What do you think?

vrabaud, 2015-09-17:
quick question: do you build all of ROS manually ? Or do you use binary packages ?
With what you're suggesting, your library would find your own OpenCV, while some catkin dependencies, would be built with a different OpenCV. That's what you want right ? If so, well, you would end up with linking with two different versions of OpenCV, which could create symbol problems (which might have worked for you because OpenCV does not break the ABI much on 2.4.x).

aleksandaratanasov, 2015-09-17:
Yeah, it's all build from source. I have only pulled the binaries that are required to do that (`wstool` for example) - just following the tutorial for Debian Wheezy. I started with the minimal ROS and started adding packages by building and installing those. As for the OpenCV - I want to have a single, custom build version of OpenCV on my system (I'm working on doing all that on Raspberry Pi so tweaking the library for this platform would be great). I have experimented with both `2.4.9` and `3.0.0`. With both I ended up with broken image_proc and a bunch of other things upon executing my nodes (building and linking was without any problems though). Mixing multiple versions of OpenCV leads exactly to such problems (including the one you've mentioned). That is why I'm trying to figure out a way of completely separating OpenCV from ROS as much as the ROS packages that use it allow that. Obviously if the source of those packages require for example something very specific that was in 2.4.9 but is no longer there in 3.0.0 there is nothing I can do except for maybe changing the code of the packages themselves which however will make the installed ROS incompatible with the one that the user will probably use to update his/her packages in the future.

I'm saying all this because officially OpenCV just like PCL is supposed to be an external library. However the way both are attached to ROS right now makes it quite difficult (impossible?) to add custom build versions to the ROS infrastructure without changing CMakeLists.txt files and whatnot.

The way I imagine it to work is:

* User builds and installs OpenCV (with all the tweaks he/she wants to add - in my case: on_nurbs, Qt support, OpenGL etc.)
* User recompiles the packages that use OpenCV (image_pipeline and so on) without the need to touch any of the files in those packages unless the version installed is not compatible with the sources in those packages.

Even with an upstream `2.4.9` on my Debian and a custom build again of `2.4.9` I need to go through a lot of trouble to set things up.

I also find it surprising that some of the CMakeLists.txt files for Indigo (hence Ubuntu 14.04 LTS) have parts that actually handle the case when OpenCV 3 is used however Ubuntu 14.04 doesn't have this version upstream.

vrabaud, 2015-09-17:
That should be the case: if your CMAKE_PREFIX_PATH includes your /usr/local (or wherever you install your custom version) and same with LD_LIBRARY_PATH, then you should be good.

If there is a bug, it could be upstream: You can try a ```find_package(OpenCV)``` in a pure CMake and see the values of ```OpenCV_INCLUDE_DIRS``` , ```OpenCV_LIBRARIES``` ```OpenCV_LIBRARY_DIRS```. That's all ROS is doing soooo ....

vrabaud, 2015-09-17:
and OpenCV 3 is packaged in ROS: the code compiles with it (and tests pass) but I could never test it live.

aleksandaratanasov, 2015-09-17:
I've just noticed (after probably the 5th or 6th rebuild of my ROS) that actually `rosdep install --from-paths src --ignore-src --rosdistro indigo -y -r --os=debian:jessie` **always** installs the `libopencv-*2.4` from upstream. Since the `wstool` is responsible for downloading all the sources I think it's safe to say that this behaviour is indeed hard-coded somewhere (I have no idea in which ROS configuration file/CMakeLists.txt the culprit is hinding). I have to mention here that I have used `checkinstall` and named my custom OpenCV build `libopencv-dev` 

  I see that `rosdep` tries to install `python-opencv` which (I've tested it) is actually the trigger for most (all?) libopencv-related upstream packages:

    sudo apt-get install python-opencv 
    [sudo] password for USER: 
    Reading package lists... Done
    Building dependency tree       
    Reading state information... Done
    The following package was automatically installed and is no longer required:
      python-wxversion
    Use 'apt-get autoremove' to remove it.
    The following extra packages will be installed:
      libopencv-calib3d2.4 libopencv-contrib2.4 libopencv-core2.4
      libopencv-features2d2.4 libopencv-flann2.4 libopencv-highgui2.4
      libopencv-imgproc2.4 libopencv-legacy2.4 libopencv-ml2.4
      libopencv-objdetect2.4 libopencv-photo2.4 libopencv-video2.4
    The following NEW packages will be installed:
      libopencv-calib3d2.4 libopencv-contrib2.4 libopencv-core2.4
      libopencv-features2d2.4 libopencv-flann2.4 libopencv-highgui2.4
      libopencv-imgproc2.4 libopencv-legacy2.4 libopencv-ml2.4
      libopencv-objdetect2.4 libopencv-photo2.4 libopencv-video2.4 python-opencv
    0 upgraded, 13 newly installed, 0 to remove and 6 not upgraded.
    Need to get 0 B/3,747 kB of archives.
    After this operation, 11.9 MB of additional disk space will be used.
    Do you want to continue? [Y/n]

  It seems that the only way of actually building from source is through a catkin workspace (or rosbuild workspace if one still uses it) by cloning into the respective repos (image_pipeline, vision_opencv etc.) and using `catkin_make` but before that looking into all the files searching for a spot where the call to OpenCV's libs and headers might get screwed up. After that of course installation is possible the usage of `rosdep` is to be avoided, which however is a great flaw in this whole thing since the `wstool` + `rosdep` combo described in the tutorial seems to be the easiest way of building from source.

  This is exactly what I don't like about the current state of things. If the include dirs and libs are moved around as I have suggested `catkin_make` will take first the custom build libs and then go for the rest.

**PS:** My LD_LIBRARY_PATH is empty (typical for Debian and Debian-based systems). However `ldconfig` has no problem pointing at my libopencv custom libraries, which - as per default - are installed in `/usr/local` (the output below is from my Debian with custom build of OpenCV installed as `libopencv-dev` as suggested by the ROS tutorial on installing ROS on Debian systems and without any of the `libopencv-*2.4` libs from upstream):

    /usr/local/lib:
	libpcl_people.so.1.7 -> libpcl_people.so.1.7.2
	liburdfdom_world.so.0.3 -> liburdfdom_world.so.0.3
	libopencv_photo.so.3.0 -> libopencv_photo.so.3.0.0 <-------------my OpenCV build
	libopencv_imgproc.so.3.0 -> libopencv_imgproc.so.3.0.0 <-------------my OpenCV build
	libconsole_bridge.so -> libconsole_bridge.so
	liburdfdom_sensor.so.0.3 -> liburdfdom_sensor.so.0.3
	libpcl_registration.so.1.7 -> libpcl_registration.so.1.7.2
	libpcl_segmentation.so.1.7 -> libpcl_segmentation.so.1.7.2
	libwx_gtk2_adv-2.8.so.0 -> libwx_gtk2_adv-2.8.so.0.8.0
	libwx_base_xml-2.8.so.0 -> libwx_base_xml-2.8.so.0.8.0
	libopencv_stitching.so.3.0 -> libopencv_stitching.so.3.0.0 <-------------my OpenCV build
	libpcl_tracking.so.1.7 -> libpcl_tracking.so.1.7.2
	libopencv_core.so.3.0 -> libopencv_core.so.3.0.0 <-------------my OpenCV build
	libwx_gtk2_html-2.8.so.0 -> libwx_gtk2_html-2.8.so.0.8.0
	libpcl_kdtree.so.1.7 -> libpcl_kdtree.so.1.7.2
	libpcl_search.so.1.7 -> libpcl_search.so.1.7.2
	liburdfdom_model.so.0.3 -> liburdfdom_model.so.0.3
	libopencv_videoio.so.3.0 -> libopencv_videoio.so.3.0.0 <-------------my OpenCV build
	libpcl_surface.so.1.7 -> libpcl_surface.so.1.7.2
	libpcl_visualization.so.1.7 -> libpcl_visualization.so.1.7.2
	libopencv_videostab.so.3.0 -> libopencv_videostab.so.3.0.0 <-------------my OpenCV build
	libpcl_io.so.1.7 -> libpcl_io.so.1.7.2
	libopencv_shape.so.3.0 -> libopencv_shape.so.3.0.0 <-------------my OpenCV build
	libwx_gtk2_aui-2.8.so.0 -> libwx_gtk2_aui-2.8.so.0.8.0
	libopencv_calib3d.so.3.0 -> libopencv_calib3d.so.3.0.0 <-------------my OpenCV build
	liburdfdom_model_state.so.0.3 -> liburdfdom_model_state.so.0.3
	libwx_gtk2_qa-2.8.so.0 -> libwx_gtk2_qa-2.8.so.0.8.0
	libopencv_superres.so.3.0 -> libopencv_superres.so.3.0.0 <-------------my OpenCV build
	libpcl_keypoints.so.1.7 -> libpcl_keypoints.so.1.7.2
	libopencv_ml.so.3.0 -> libopencv_ml.so.3.0.0 <-------------my OpenCV build
	libopencv_viz.so.3.0 -> libopencv_viz.so.3.0.0 <-------------my OpenCV build
	libwx_base-2.8.so.0 -> libwx_base-2.8.so.0.8.0
	libpcl_filters.so.1.7 -> libpcl_filters.so.1.7.2
	libwx_gtk2_richtext-2.8.so.0 -> libwx_gtk2_richtext-2.8.so.0.8.0
	libcollada-dom2.4-dp.so.0 -> libcollada-dom2.4-dp.so.2.4.0
	libwx_gtk2_core-2.8.so.0 -> libwx_gtk2_core-2.8.so.0.8.0
[... cut ...]
	libwx_gtk2_xrc-2.8.so.0 -> libwx_gtk2_xrc-2.8.so.0.8.0

`pkg-config` is also using without any issues:

    pkg-config --cflags opencv
      -I/usr/local/include/opencv -I/usr/local/include 

and 

    pkg-config --cflags opencv
      -L/usr/local/lib -lopencv_shape -lopencv_stitching -lopencv_objdetect -lopencv_superres -lopencv_videostab -lopencv_calib3d -lopencv_features2d -lopencv_highgui -lopencv_videoio -lopencv_imgcodecs -lopencv_video -lopencv_photo -lopencv_ml -lopencv_imgproc -lopencv_flann -lopencv_viz -lopencv_core -lopencv_hal

vrabaud, 2015-09-18:
before moving further, ```catkin_make``` is just a stupid CMake invocation. So, when it tries to find OpenCV, it just does a ```find_package(OpenCV)```. Is your custom share folder where you have a OpenCV/OpenCVConfig.cmake on your ```CMAKE_PREFIX_PATH``` ?
Rules are explained here: http://www.cmake.org/cmake/help/v3.0/command/find_package.html

aleksandaratanasov, 2015-09-18:
Yeah, I know that catkin_make is a wrapper for CMake. As for the **cmake** file:

    user@debian:/usr/local/share/OpenCV$ ls
    haarcascades  OpenCVConfig.cmake          OpenCVModules.cmake
    lbpcascades   OpenCVConfig-version.cmake  OpenCVModules-release.cmake

As I've mentioned: even without any upstream installation and only my custom OpenCV installed rosdep ignores all that an starts downloading and installing upstream packages. If someone points me where I should start looking to find out why that is, I would gladly investigate further.

vrabaud, 2015-09-21:
well, you mention several issues. Concerning rosdep, it will look for a package.xml with libopencv-dev. If it does not find it, it will look in the rosdep file for the macthing key which in your case is: https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L1729

So either have OpenCV in your workspace with a package.xml, or do not use rosdep (or use it but uninstall libopencv-dev afterwards).

Now, concerning using a custom OpenCV: either you have it in your workspace (with a package.xml), either you build it and install it in a custom location (which is what you seem to have right now). To find it, you then need to complete your ```CMAKE_PREFIX_PATH```. Please write a simple CMake project that only does:
```
cmake_minimum_required(VERSION 2.8)
project(test_opecv)
find_package(OpenCV)
message(${OpenCV_INCLUDE_DIRS})
message(${OpenCV_LIBRARIES})
```
and get it to spit out your right path (in ```/usr/local``` I guess) by modifying ```CMAKE_PREFIX_PATH```

aleksandaratanasov, 2015-09-21:
I don't have any issues with my custom build of OpenCV in general (hence that thing with the minimal example is unnecessary since I've already tested it). I'm just having difficulties building ROS packages that depend on OpenCV. As for the `base.yaml` file:

    libopencv-dev:
      arch: [opencv]
      debian: [libopencv-dev] <------- this is how I installed my custom build using the tutorial for Debian installation for ROS
      fedora: [opencv-devel]
      gentoo: [media-libs/opencv]
      macports: [opencv]
      ubuntu:
        saucy: [libopencv-dev]
        trusty: [libopencv-dev]
        utopic: [libopencv-dev]
        vivid: [libopencv-dev]

As I said rosdep ignores my custom installation **libopencv-dev** and goes on with installing the upstream packages. Either the YAML configuraiton file is incorrect or the tutorial is faulty, or rosdep is doing something it isn't supposed to. My custom build of OpenCV is in /usr/local, I can build applications with `cmake` without a problem...Unless they are ROS-related.

vrabaud, 2015-09-21:
wait, something that I totally missed then: what you call your custom ```libopencv-dev``` is actually a custom package that you built and then installed with ```dpkg -i``` ? It's not sources you compiled and installed with a ```make install``` ?

vrabaud, 2015-09-21:
otherwise, this is how rosdep installs things on debian:
https://github.com/ros-infrastructure/rosdep/blob/master/src/rosdep2/platforms/debian.py#L119

aleksandaratanasov, 2015-09-21:
Here is what I did (straight from the tutorial):

1. Cloned OpenCV 3.0.0
2. Created a build directory inside the cloned sources root directory
3. Ran `cmake ..` to create the CMakeCache etc.
4. Ran `cmake-gui ..` to configure my OpenCV build - enabled OpenGL, Matlab (phew, it took me a while which paths I had to add from my Matlab R2014a installation)
5. Ran `make`
6. Ran `sudo checkinstall make install` - renamed the package to `libopencv-dev` (as per tutorial and also as you can see in the `base.yaml`), added description, version (same as the one in Debian's repos) and license

Note I have tried both with `checkinstall` (and renaming the package as required) and `make install` in both cases `rosdep` ignores all this. In the later of course it's much tedious to clean up afterwards since the package manager doesn't keep track of the install hence you have to manually delete stuff.

aleksandaratanasov, 2015-09-21:
Small update: I managed to fully install the `image_pipeline`, `image_common` and `vision_opencv` by building the required sources in my `catkin` workspace and then using `catkin_make install` to install all the files to my `/opt/ros/indigo`. It seems that for now this is the way to go. `rosdep` seems to be handling correctly some packages so at least for the barebone installation of ROS from source it is worth using but not when it comes to the `image_pipeline`, `image_common` and `vision_opencv` in case a custom build of OpenCV is installed on the system.

vrabaud, 2016-01-10:
For OpenCV 3 support, you need to build it so that it does not appear automatically on your different paths, cf
http://wiki.ros.org/vision_opencv#OpenCV3
OpenCV 3.1 also contains CMake fixes that are helpful for that configuration.

closed 2016-01-10 by vrabaud

SimonBirrell, 2016-01-30:
I'm having the same or similar issue. I have a custom install of OpenCV (with CUDA support) and I can compile anything that doesn't involve ROS.

catkin_make gives an error

    make[2]: *** No rule to make target `/usr/lib/arm-linux-gnueabihf/libopencv_videostab.so.2.4.8', needed by `/home/ubuntu/catkin_ws/devel/lib/ros_caffe/ros_caffe_test'.  Stop.

whereas the custom version is 2.4.12. Printing out OpenCV_VERSION from the CmakeFiles.txt file gives

    -- OpenCV_VERSION: 2.4.12.2

I've tried aleksandaratanasov's suggestion of adding vision_opencv, image_pipeline and image_common to the Catkin workspace, but then I enter dependency hell. It starts asking for geometry_experimental, tf2, bullet etc. etc.  

Surely it's not necessary to recompile half of ROS just to get it linking with the desired version of OpenCV ????

OAkyildiz, 2020-07-27:
The year 2020, and this still has a similar behavior. Why doesn't ROS have the option to link against another build of OpenCV instead of the one provided by vision_opencv?

JWhitleyWork, 2020-07-27:
@OAkyildiz I'm sorry but this really isn't something that `image_pipeline` has control over. Please take this up with https://github.com/ros-perception/vision_opencv/issues