ros-perception/image_pipeline #152: image view freezes with opencv 3

ipa-bnm, 2015-10-07:
when i use image view with opencv3 (3.0.0-2) from shadow fixed, image view freezes before showing any image. I've installed ros-indigo-opencv3 3.0.0-2 from shadow fixed and overlayed image_pipeline + vision_opencv from source 
What i've found out so far:
- ros-indigo-opencv3 is compiled with QT support instead of Gtk which forced me to remove the HAVE_GTK definition inside this [CMakeLists](https://github.com/ros-perception/image_pipeline/blob/indigo/image_view/CMakeLists.txt#L45)
- now image view freezes on cv::imshow() call
- i could find out, that image view starts working again if i do not use named Windows and add an waitKey call. So i commented this [lines](https://github.com/ros-perception/image_pipeline/blob/indigo/image_view/src/nodelets/image_nodelet.cpp#L136-L137) and added an cv::waitKey(1) [here](https://github.com/ros-perception/image_pipeline/blob/indigo/image_view/src/nodelets/image_nodelet.cpp#L188).

This leads me to the conclusion that namedWindows and startWindowThread is not working with OpenCV 3 and QT, is this correct? Did someone experienced the same problem?

ipa-bnm, 2015-10-07:
@dejanpan

BenBlumer, 2015-10-27:
I think I'm getting a similar problem.

`rosrun image_view image_view` 

yields:

```
[ INFO] [1445983450.803787659]: Using transport "raw"
OpenCV Error: Unspecified error (The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Carbon support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script) in cvNamedWindow, file /tmp/buildd/ros-indigo-opencv3-3.0.0-1trusty-20150620-0953/modules/highgui/src/window.cpp, line 517

OpenCV Error: Unspecified error (The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Carbon support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script) in cvDestroyWindow, file /tmp/buildd/ros-indigo-opencv3-3.0.0-1trusty-20150620-0953/modules/highgui/src/window.cpp, line 523
terminate called after throwing an instance of 'cv::Exception'
  what():  /tmp/buildd/ros-indigo-opencv3-3.0.0-1trusty-20150620-0953/modules/highgui/src/window.cpp:523: error: (-2) The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Carbon support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script in function cvDestroyWindow
```

meyerj, 2018-11-15:
I rebuilt `ros-kinetic-opencv3` from https://github.com/ros-gbp/opencv3-release/tree/release/kinetic/opencv3 and disabled QT using the `-DUSE_QT=OFF` flag. This fixes the issue. The official Ubuntu package for opencv3 also does not build with Qt and uses GTK3 instead:
https://salsa.debian.org/science-team/opencv/tree/master/

JWhitleyAStuff, 2019-11-25:
This is a very old ticket but many changes have been made to `image_view` since then. If #481 is not yet merged, try the branch `fix/reliably-close-image-view`. If that PR is merged, try the `melodic` branch. Please re-open the ticket if the issue is not solved.

closed 2019-11-25 by JWhitleyAStuff