ros-perception/image_pipeline #153: image_view: Attempt to unlock mutex that was not locked

joseparnau, 2015-10-16:
Hi,

I tried to visualize an image captured from the parrot ardrone 2 using the image_view node and i got:
--------------------------------------------
$ rosrun image_view image_view image:=/ardrone/front/image_raw
[ INFO] [1445004668.853536140]: Using transport "raw"
[ INFO] [1445004669.204656248]: Press 'q' on the window to stop the node
Attempt to unlock mutex that was not locked
Aborted
---------------------------------------------

There was this issue:
https://github.com/ros-perception/image_pipeline/issues/106
but seems it was fixed.

Any clue?
My ROS version is Jade on debian.

Thank's in advance.

joseparnau, 2015-10-19:
It works with Ubuntu Jade so it seems that Debian is not a good choice for that node...

joseparnau, 2015-10-21:
Hi,
It looks like the error might be due to my own installation. You can close the thread.

closed 2015-10-21 by joseparnau

vrabaud, 2015-10-21:
if you ever figure out, please let me know. Thx