ros-perception/image_pipeline #154: distortion coefficients > 3 not working

anuppari, 2015-10-26:
Setting the requested number of distortion coefficients using either the `-k` or `--k-coefficients` flags to any value greater than 3 results in the output distortion coefficients being set to all zeros. My exact command line argument is:

    rosrun camera_calibration cameracalibrator.py --size 8x6 --square 0.06 -k 4 image:=/mv_BF001066/image_raw camera:=/mv_BF001066

which returns

    camera matrix
    556.831101 0.000000 366.924119
    0.000000 553.420218 253.218759
    0.000000 0.000000 1.000000
    
    distortion
    0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000

However running:

    rosrun camera_calibration cameracalibrator.py --size 8x6 --square 0.06 -k 3 image:=/mv_BF001066/image_raw camera:=/mv_BF001066

results in

    camera matrix
    554.412673 0.000000 364.608596
    0.000000 551.032409 252.415234
    0.000000 0.000000 1.000000
    
    distortion
    -0.404856 0.266958 0.001336 0.000190 -0.130004

plusk01, 2017-09-13:
I am seeing the same thing. Any status updates on this?

DavidTorresOcana, 2018-08-14:
I am getting the same. Any update?

SteveMacenski, 2019-11-04:
PRs welcome

DavidTorresOcana, 2019-11-05:
I was not able to reproduce this problem with a pinhole camera.
This **may** happen when trying to use Rational polynomial distortion model on camera with large distortion like fisheye lenses. This is an issue with the geometry and OpenCV optimization itself, not this package.

Just a clarification on ```camera_calibration``` tool flags:
 * if ```-k ``` or ```--k-coefficients``` is set to 3 or less, ```PLUMB_BOB``` model is used
 *  if ```-k ``` or ```--k-coefficients``` is set to 4 or more, ```RATIONAL_POLYNOMIAL``` model is used

It may be worth put it into the wiki

SteveMacenski, 2019-11-06:
This ticket is also 4 years old. Its possible its been fixed since then. 

Closing.

(Also welcome to update the wiki, you can probably explain it better than I. I make my interns calibrate cameras for me)

closed 2019-11-06 by SteveMacenski