ros-perception/image_pipeline #155: image_proc crashes

Georacer, 2015-10-27:
Hello everyone,

I'm using this code in my ROS launch file to raise the `image_proc` node:

   <node pkg="image_proc" name="image_proc" type="image_proc" output="screen"/>

I'm publishing an `image_raw` topic and the corresponding `camera_info` topic in the same namespace.

I fill `camera_info` topic with this info, coming form the calibration: 

    # oST version 5.0 parameters

	[image]

	width
	4000

	height
	3000

	[narrow_stereo]

	camera matrix
	3539.654538 0.000000 1980.239959
	0.000000 3530.648540 1482.550254
	0.000000 0.000000 1.000000

	distortion
	0.000012 -0.154715 -0.006182 -0.001827 0.000000

	rectification
	1.000000 0.000000 0.000000
	0.000000 1.000000 0.000000
	0.000000 0.000000 1.000000

	projection
	3479.307129 0.000000 1971.273888 0.000000
	0.000000 3510.014893 1469.491590 0.000000
	0.000000 0.000000 1.000000 0.000000

which I have first converted into a yaml file and loaded it into the parameter server:

    image_width: 4000
    image_height: 3000
    camera_name: multispek
    camera_matrix:
      rows: 3
      cols: 3
      data: [3539.654538, 0.0, 1980.239959, 0.0, 3530.64854, 1482.550254, 0.0, 0.0, 1.0]
    distortion_model: plumb_bob
    distortion_coefficients:
      rows: 1
      cols: 5
      data: [1.2e-05, -0.154715, -0.006182, -0.001827, 0.0]
    rectification_matrix:
      rows: 3
      cols: 3
      data: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
    projection_matrix:
      rows: 3
      cols: 4
      data: [3479.307129, 0.0, 1971.273888, 0.0, 0.0, 3510.014893, 1469.49159, 0.0, 0.0, 0.0, 1.0, 0.0]

The problem is that **occasionally, with the same picture**, image_proc crashes with a message such as this. The number of "OpenCV Error" lines varies.

    OpenCV Error: Assertion failed (cn <= 4 && ssize.area() > 0) in remapBilinear, file /build/buildd/opencv-2.4.8+dfsg1/modules/imgproc/src/imgwarp.cpp, line 2425
    OpenCV Error: Assertion failed (cn <= 4 && ssize.area() > 0) in remapBilinear, file /build/buildd/opencv-2.4.8+dfsg1/modules/imgproc/src/imgwarp.cpp, line 2425
    OpenCV Error: Assertion failed (cn <= 4 && ssize.area() > 0) in remapBilinear, file /build/buildd/opencv-2.4.8+dfsg1/modules/imgproc/src/imgwarp.cpp, line 2425
    OpenCV Error: Assertion failed (cn <= 4 && ssize.area() > 0) in remapBilinear, file /build/buildd/opencv-2.4.8+dfsg1/modules/imgproc/src/imgwarp.cpp, line 2425
    OpenCV Error: Assertion failed (cn <= 4 && ssize.area() > 0) in remapBilinear, file /build/buildd/opencv-2.4.8+dfsg1/modules/imgproc/src/imgwarp.cpp, line 2425
    OpenCV Error: Assertion failed (cn <= 4 && ssize.area() > 0) in remapBilinear, file /build/buildd/opencv-2.4.8+dfsg1/modules/imgproc/src/imgwarp.cpp, line 2425
    OpenCV Error: Assertion failed (cn <= 4 && ssize.area() > 0) in remapBilinear, file /build/buildd/opencv-2.4.8+dfsg1/modules/imgproc/src/imgwarp.cpp, line 2425
    OpenCV Error: Assertion failed (cn <= 4 && ssize.area() > 0) in remapBilinear, file /build/buildd/opencv-2.4.8+dfsg1/modules/imgproc/src/imgwarp.cpp, line 2425
    terminate called after throwing an instance of 'tbb::captured_exception'
      what():  /build/buildd/opencv-2.4.8+dfsg1/modules/imgproc/src/imgwarp.cpp:2425: error: (-215) cn <= 4 && ssize.area() > 0 in function remapBilinear

    [camera/image_proc-5] process has died [pid 25007, exit code -6, cmd /opt/ros/indigo/lib/image_proc/image_proc __name:=image_proc __log:=/home/george/.ros/log/9dca421c-7ce4-11e5-951c-40f02fa53f38/camera-image_proc-5.log].
    log file: /home/george/.ros/log/9dca421c-7ce4-11e5-951c-40f02fa53f38/camera-image_proc-5*.log

I repeat, I may load the same picture sometime and it may or may not crash.

Is it possible that I have a badly formed calibration file which causes transformation singularities?

Thanks in advance,
George

dejunzhang, 2017-02-16:
hi, @Georacer , did you soloved your problem? I met the same problem and don't know how to resolove ti.

Georacer, 2017-02-16:
Oh boy... I'm afraid this case has gone REALLY cold.

I'm afraid I don't remember exactly if and how I resolved this issue and this code was part of a project *I don't want to touch again*. Sorry for the emphasized disgust.

So I can't be of much help, except for going through my blurry memories (and a bit of browsing of that code).
According to these, I didn't manage to solve the problem, based on the fact that image_proc is still commented out in my launch file.

What I think I did was do the image processing on the distorted image, for point extraction and then use the image calibration configuration in the Python code which was doing geolocation, to feed an _image_geometry_ object which did the rectification in a coordinate level.

Essentially, instead of rectifying the image and then process it, I processed the image first and rectified the resulting coordinates.

But of course this may not be doable in your application.

2ros0, 2018-02-28:
I have the same problem. Does anyone have a solution?

SteveMacenski, 2019-10-16:
Been cold for a few years and sounds like there's a reasonable work around

closed 2019-10-16 by SteveMacenski