ros-perception/image_pipeline #156: point_cloud2 nodelet works only with image_rect_color

trigal, 2015-10-30:
The nodelet point_cloud2.cpp subscribes to the left/image_rect_color, it is hard-coded, even if it handles also MONO8 images. I think this should be a parameter of the node, like the others defined under the cfg folder. 
For my needs I recompiled the code changing 

    ros::NodeHandle &nh = getNodeHandle();
    // Queue size 1 should be OK; the one that matters is the synchronizer queue size.
    image_transport::TransportHints hints("raw", ros::TransportHints(), getPrivateNodeHandle());
    sub_l_image_  .subscribe(*it_, "left/image_rect_color", 1, hints);
    sub_l_info_   .subscribe(nh,   "left/camera_info", 1);
    sub_r_info_   .subscribe(nh,   "right/camera_info", 1);
    sub_disparity_.subscribe(nh,   "disparity", 1);

into

    ros::NodeHandle &nh = getNodeHandle();
    // Queue size 1 should be OK; the one that matters is the synchronizer queue size.
    image_transport::TransportHints hints("raw", ros::TransportHints(), getPrivateNodeHandle());
    sub_l_image_  .subscribe(*it_, "left/image_rect", 1, hints);
    sub_l_info_   .subscribe(nh,   "left/camera_info", 1);
    sub_r_info_   .subscribe(nh,   "right/camera_info", 1);
    sub_disparity_.subscribe(nh,   "disparity", 1);

I think it is worth to add that one :-)

vrabaud, 2015-11-15:
well, you can just remap the topic right ? For the mono8 support, it should be added to the doc http://wiki.ros.org/stereo_image_proc , that's it right ?

SteveMacenski, 2019-11-03:
VIncent's comment is correct, closing

closed 2019-11-03 by SteveMacenski