ros-perception/image_pipeline #176: Compiling a ROS/OpenCV application tries to link against wrong version of OpenCV

SimonBirrell, 2016-01-30:
I'm trying to use OpenCV with ROS Indigo on an ARM-based Jetson TK1 board. The board ships with it's own version of OpenCV 2 (OpenCV4Tegra) which works just fine. I can compile simple OpenCV applications with cmake / make and everything links and executes correctly.

But when I try to compile a ROS/OpenCV application (in this case ros_caffe) with catkin_make, I get the following error:

    ####
    #### Running command: "make -j1 -l1" in "/home/ubuntu/catkin_ws/build"
    ####
    Scanning dependencies of target ros_caffe_test
    [100%] Building CXX object ros_caffe/CMakeFiles/ros_caffe_test.dir/src/ros_caffe_test.cpp.o
    make[2]: *** No rule to make target `/usr/lib/arm-linux-gnueabihf/libopencv_videostab.so.2.4.8', needed by `/home/ubuntu/catkin_ws/devel/lib/ros_caffe/ros_caffe_test'.  Stop.
    make[1]: *** [ros_caffe/CMakeFiles/ros_caffe_test.dir/all] Error 2
    make: *** [all] Error 2
    Invoking "make -j1 -l1" failed

The OpenCV2 that ships with the board is 2.4.12.2 not 2.4.8. Strangely enough, if I insert the following into CMakeFiles.txt of ros_caffe:

    MESSAGE( STATUS "OpenCV_INCLUDE_DIRS: " ${OpenCV_INCLUDE_DIRS} )
    MESSAGE( STATUS "OpenCV_LIBRARIES: " ${OpenCV_LIBRARIES} )
    MESSAGE( STATUS "OpenCV_LIBRARY_DIRS: " ${OpenCV_LIBRARY_DIRS} )
    MESSAGE( STATUS "OpenCV_VERSION: " ${OpenCV_VERSION} )

I get:

    -- OpenCV_INCLUDE_DIRS: /usr/include/opencv/usr/include
    -- OpenCV_LIBRARIES: opencv_vstabopencv_tegraopencv_imuvstabopencv_facedetectopencv_esm_panoramaopencv_detection_based_trackeropencv_videostabopencv_videoopencv_superresopencv_stitchingopencv_photoopencv_objdetectopencv_mlopencv_legacyopencv_imgprocopencv_highguiopencv_gpuopencv_flannopencv_features2dopencv_coreopencv_contribopencv_calib3d
    -- OpenCV_LIBRARY_DIRS: 
    -- OpenCV_VERSION: 2.4.12.2

which looks good.

How can I force catkin/cmake to compile against 2.4.12.2 and not 2.4.8? The target library appears to be in /usr/lib:

    libopencv_videostab.so
    libopencv_videostab.so.2.4
    libopencv_videostab.so.2.4.12

I've tried a suggestion from aleksandaratanasov issue 149 to recompile vision_opencv, image_pipeline and image_common in the catkin workspace, but that seems to lead me down an ever-deeper rabbit hole of hard-to-install dependencies like bullet.

There must be some way of telling catkin/cmake to link against a specific version of OpenCV2 ?

Thanks.

vrabaud, 2016-01-30:
where does your 2.4.8 come from ?
ROS will take whatever is first on your path so you probably have an error there. 
You can force a specific version with the version field in CMake otherwise, https://cmake.org/cmake/help/v3.0/command/find_package.html

SimonBirrell, 2016-01-31:
Setting the version in find_package makes no difference. find_package was already locating the correct version 2.4.12.2 (see OpenCV_VERSION) above. The issue seems to occur later.

Doing a bit of grepping for "opencv" in /opt/ros I get many explicit references to 2.4.8, for example:

    indigo/share/pano_core/cmake/pano_coreConfig.cmake:set(libraries "pano_core;/usr/lib/arm-linux-gnueabihf/libboost_program_options.so;/usr/lib/arm-linux-gnueabihf/libopencv_videostab.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_video.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_superres.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_stitching.so.2.4.8;/usr/lib/arm-linux-gnueabihf/liopencv_photo.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_ocl.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_objdetect.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_ml.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_legacy.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_imgproc.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_highgui.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_gpu.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_flann.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_features2d.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_core.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_contrib.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_calib3d.so.2.4.8")

    indigo/share/cv_bridge/cmake/cv_bridgeConfig.cmake:set(libraries "cv_bridge;/usr/lib/arm-linux-gnueabihf/libopencv_videostab.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_video.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_superres.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_stitching.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_photo.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_ocl.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_objdetect.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_ml.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_legacy.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_imgproc.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_highgui.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_gpu.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_flann.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_features2d.so.2.4.8;/usr/lib/arm-linux-gnueabihf/liopencv_core.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_contrib.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_calib3d.so.2.4.8")

    indigo/share/compressed_image_transport/cmake/compressed_image_transportConfig.cmake:set(libraries "compressed_image_transport;/usr/lib/arm-linux-gnueabihf/libopencv_videostab.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_video.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_ts.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_superres.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_stitching.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_photo.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_ocl.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_objdetect.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_ml.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_legacy.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_imgproc.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_highgui.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_gpu.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_flann.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_features2d.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_core.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_contrib.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_calib3d.so.2.4.8")

    indigo/share/image_proc/cmake/image_procConfig.cmake:set(libraries "image_proc;/usr/lib/arm-linux-gnueabihf/libopencv_videostab.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_video.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_superres.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_stitching.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_photo.so.2.4.8;/usr/lib/arm-linux-gnueabihf/liopencv_ocl.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_objdetect.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_ml.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_legacy.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_imgproc.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_highgui.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_gpu.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_flann.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_features2d.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_core.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_contrib.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_calib3d.so.2.4.8")

    indigo/share/pano_py/cmake/pano_pyConfig.cmake:set(libraries "pano_cv;pano_py;/usr/lib/arm-linux-gnueabihf/libboost_python.so;/usr/lib/arm-linux-gnueabihf/libopencv_videostab.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_video.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_superres.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_stitching.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_photo.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_ocl.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_objdetect.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_ml.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_legacy.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_imgproc.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_highgui.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_gpu.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_flann.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_features2d.so.2.4.8;/usr/lib/arm-linux-gnueabihf/libopencv_core.so.2.4.8;/usr/lib/ar [... cut ...] linux-gnueabihf/libopencv_highgui.so.2.4.8 -l:/usr/lib/arm-linux-gnueabihf/libopencv_gpu.so.2.4.8 -l:/usr/lib/arm-linux-gnueabihf/libopencv_flann.so.2.4.8 -l:/usr/lib/arm-linux-gnueabihf/libopencv_features2d.so.2.4.8 -l:/usr/lib/arm-linux-gnueabihf/libopencv_core.so.2.4.8 -l:/usr/lib/arm-linux-gnueabihf/libopencv_contrib.so.2.4.8 -l:/usr/lib/arm-linux-gnueabihf/libopencv_calib3d.so.2.4.8

Maybe these files were created when ROS was installed. But they sure like the source of the "2.4.8" problem.

vrabaud, 2016-01-31:
You still did not explain where 2.4.8 comes from on your machine. I am no ARM expert.
From what I can gather you have an Ubuntu ARM on your TK1 which includes both NVidia's 2.4.12 OpenCV and Ubuntu's official 2.4.8 ? Is that the case ?

trainman419, 2016-01-31:
It sounds like you're using the precompiled versions of the ROS image_pipeline and vision_opencv packages from the deb repo. These packages are built against standard Ubuntu versions of OpenCV, and can't (or shouldn't) be used with CV for Tegra.

If you have the Ubuntu opencv packages (or a shim) installed, you should remove it and build vision_opencv, image_pipeline and image_common from source in your catkin workspace. If you're missing build dependencies, you can install them using rosdep.

trainman419, 2016-01-31:
You might have followed [this ROS answer](http://answers.ros.org/question/196194/ros-indigo-installation-for-ubuntu-arm-jetson-tk1-ubuntu-1404/?answer=196197#post-id-196197) If you did, it looks like Nvidia's version of OpenCV4Tegra is no longer compatible. You should uninstall the `libopencv-dev-dummy` package and build vision_opencv, image_pipeline and image_common from source.

SimonBirrell, 2016-01-31:
Thanks for all the suggestions. I think trainman419 has identified the problem. The pre-installed version of ROS that comes no the Jetson TK1 (actually, on the "Autonomous Deep Learning Robot") is likely from the deb repo and so comes precompiled for 2.4.8.

I abandoned the recompilation of vision_opencv because I got stuck installing bullet, one of the dependencies. The suggestion of using rosdep might have helped here, but by then I'd found a long winded solution as follows:

What I've ended up doing, which works but is a pain, is to manually edit all those .cmake and .pc files and remove the references to 2.4.8 and change the directory the libraries are supposed to be in. It took me a while, but after doing that I could compile the package I was working on. I also removed references to libopencv_ocl.so which appears to have disappeared between 2.4.8 and 2.4.12.

Now I have a ROS-powered robot with OpenCV, Caffe and object recognition. I'm happy again! If anyone's interested in the details see [http://www.artificialhumancompanions.com/integrating-ros-caffe-opencv-on-the-autonomous-deep-learning-robot/](this post).

Thanks again for the help!

vrabaud, 2016-02-03:
ok, closing for now: there are small things that could be done like renaming /usr/lib/x86_64-linux-gnu/libopencv_videostab.so.2.4.8 in /usr/lib/x86_64-linux-gnu/libopencv_videostab.so and you could then create a symlink of the folder but let's stick to the policy of only being compatible with Ubuntu's version.
Too many bugs could arise otherwise (a simple one could be a dependency on a module that Ubuntu does not include).

closed 2016-02-03 by vrabaud

hongwenxie, 2016-05-20:
cv_bridge is the converter between ROS Image messages and OpenCV images. The default opencv version used by cv_bridge package is 2.4.8, if you installed a newer opencv version, you should re-compile the cv_bridge package.

git clone https://github.com/ros-perception/vision_opencv.git

move the cv_bridge package to catkin workspace,then:

catkin_make --pkg cv_bridge

openourweb, 2016-05-28:
Dear  hongwenxie,
   I have the same issue, I try your command, but the errors are the same. what else should I do?
Thanks
John

Abbyls, 2016-10-16:
I solved the problem as followed the advice from @hongwenxie ! Thanks!
But if i want to use the cv_bridge package in another workspace i need re-compile the cv_bridge package again! i think it's complicated. I wonder its there any other way to convenient it!
@hongwenxie