ros-perception/image_pipeline #179: Error while building image_proc

avalada, 2016-02-19:
Using the latest pull as of 19/02

`/home/foo/software/ros_catkin_ws/build_isolated/image_proc && /opt/ros/indigo/env.sh make -j8 -l8
[ 90%] Built target image_proc_gencfg
[ 90%] Building CXX object CMakeFiles/image_proc.dir/src/nodelets/crop_non_zero.cpp.o
/home/foo/software/ros_catkin_ws/src/image_pipeline/image_proc/src/nodelets/crop_non_zero.cpp:42:30: error: ‘image_encodings’ is not a namespace-name
 namespace enc = sensor_msgs::image_encodings;
                              ^
/home/foo/software/ros_catkin_ws/src/image_pipeline/image_proc/src/nodelets/crop_non_zero.cpp:42:45: error: expected namespace-name before ‘;’ token
 namespace enc = sensor_msgs::image_encodings;
                                             ^
/home/foo/software/ros_catkin_ws/src/image_pipeline/image_proc/src/nodelets/crop_non_zero.cpp: In member function ‘void image_proc::CropNonZeroNodelet::imageCb(const ImageConstPtr&)’:
/home/foo/software/ros_catkin_ws/src/image_pipeline/image_proc/src/nodelets/crop_non_zero.cpp:102:19: error: ‘sensor_msgs::image_encodings’ has not been declared
   if(sensor_msgs::image_encodings::numChannels(raw_msg->encoding) != 1){
                   ^
/home/foo/software/ros_catkin_ws/src/image_pipeline/image_proc/src/nodelets/crop_non_zero.cpp:110:28: error: ‘enc’ has not been declared
   if (raw_msg->encoding == enc::TYPE_8UC1){
                            ^
/home/foo/software/ros_catkin_ws/src/image_pipeline/image_proc/src/nodelets/crop_non_zero.cpp:120:3: error: ‘findContours’ is not a member of ‘cv’
   cv::findContours(m, cnt, CV_RETR_EXTERNAL, CV_CHAIN_APPROX_NONE);
   ^
/home/foo/software/ros_catkin_ws/src/image_pipeline/image_proc/src/nodelets/crop_non_zero.cpp:120:28: error: ‘CV_RETR_EXTERNAL’ was not declared in this scope
   cv::findContours(m, cnt, CV_RETR_EXTERNAL, CV_CHAIN_APPROX_NONE);
                            ^
/home/foo/software/ros_catkin_ws/src/image_pipeline/image_proc/src/nodelets/crop_non_zero.cpp:120:46: error: ‘CV_CHAIN_APPROX_NONE’ was not declared in this scope
   cv::findContours(m, cnt, CV_RETR_EXTERNAL, CV_CHAIN_APPROX_NONE);
                                              ^
/home/foo/software/ros_catkin_ws/src/image_pipeline/image_proc/src/nodelets/crop_non_zero.cpp:132:16: error: ‘boundingRect’ is not a member of ‘cv’
   cv::Rect r = cv::boundingRect(cnt[std::distance(cnt.begin(), it)]);
                ^
make[2]: *** [CMakeFiles/image_proc.dir/src/nodelets/crop_non_zero.cpp.o] Error 1
make[1]: *** [CMakeFiles/image_proc.dir/all] Error 2
make: *** [all] Error 2`

vrabaud, 2016-02-19:
you probably have a custom OpenCV installed no ?

avalada, 2016-02-19:
That is correct. I am using opencv 2.4

vrabaud, 2016-02-19:
which OS, compiler, architecture ? The first error is about sensor_msgs so make sure it's installed.

avalada, 2016-02-19:
Strange because I have sensor_msgs installed
Ubuntu 14.04
gcc (Ubuntu 4.9.3-8ubuntu2~14.04) 4.9.3
amd64

avalada, 2016-02-21:
Still getting the same error after removing devel_isolated and build_isolated

This is the command Im running to build
`sudo ./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release -DCUDA_TOOLKIT_ROOT_DIR=/usr/local/cuda-7.0 --install-space /opt/ros/indigo`

vrabaud, 2016-02-21:
hmmm, try to get more info with ```make VERBOSE=1```.
The ```sensor_msgs/image_encodings.h``` header is included through ```cv_bridge/cv_bridge.h``` at the top of the file. You probably have customization in your setup.

vrabaud, 2016-02-21:
It seems you are overwriting the ROS install, that is pretty bad. You probably want to overwrite the vision libs so that they include your OpenCV and cuda right ? The problem is you are also overwriting the /opt/ros/indigo/setup.* files. I am afraid you will have to reinstall your ROS.
Then do the same instruction but install to whatever like /opt/ros/indigo_local. And source the setup file from that folder when you want to use your libraries.

avalada, 2016-02-21:
Ya that was the problem. 

Thanks for the help!

closed 2016-02-21 by avalada