ros-perception/image_pipeline #180: Need to check if all distortion coefficients are zero, not just first one

lucasw, 2016-02-29:
https://github.com/ros-perception/image_pipeline/blob/indigo/image_proc/src/nodelets/rectify.cpp#L125

    // If zero distortion, just pass the message along
    if (info_msg->D.empty() || info_msg->D[0] == 0.0)
    {
    ...

That check ought to test every element of info_msg->D to see if they are all 0.0.  This check probably always works with real camera calibrations, but synthetic ones (which could be used in tests or simulations) can have an exactly zero first coefficient and non-zero other coefficients.

lucasw, 2016-02-29:
I can provide a pull request in a day or two.

vrabaud, 2016-02-29:
Thx, please do. And please include a test.

lucasw, 2016-03-01:
A corresponding change is needed in vision_opencv to make a fix here work:

https://github.com/ros-perception/vision_opencv/blob/indigo/image_geometry/src/pinhole_camera_model.cpp#L130

    cache_->distortion_state = (cam_info_.D.size() == 0 || (cam_info_.D[0] == 0.0)) ? NONE : CALIBRATED;

vrabaud, 2016-03-13:
greatly fixed in #117

closed 2016-03-13 by vrabaud