ros-perception/image_pipeline #185: [image_view2] Build error in some cases because of not found ImageViewConfig.h

wkentaro, 2016-03-13:
Firstly reported at: https://github.com/ros-perception/image_pipeline/commit/6aa8062e2635306be3322f611e010ea872535346#commitcomment-16596255
```
Scanning dependencies of target image_view_exe
[ 70%] Building CXX object CMakeFiles/image_view_exe.dir/src/nodes/image_view.cpp.o
/tmp/yaourt-tmp-username/aur-ros-indigo-image-view/src/image_view/src/nodes/image_view.cpp:34:40: fatal error: image_view/ImageViewConfig.h: No such file or directory
compilation terminated.
```

wkentaro, 2016-03-13:
@vrabaud Does this issue happen only on ArchLinux?

vrabaud, 2016-03-13:
well, I was not able to reproduce it anywhere else, that's why I renamed that issue. But feel free to change as it seems you could reproduce.

vrabaud, 2016-03-13:
Please close if issue is fixed with master.

wkentaro, 2016-03-13:
> well, I was not able to reproduce it anywhere else, that's why I renamed that issue. But feel free to change as it seems you could reproduce.

I see. In my environment I cannot reproduce, either.
@nickoe Please check if this issue is fixed in your environment.

nicolaerosia, 2016-03-13:
I had the same issue on Fedora 23.
With the latest commits it seems to be gone.

vrabaud, 2016-03-14:
ok, well, closing for now then. Please re-open if you can reproduce.

closed 2016-03-14 by vrabaud

nickoe, 2016-03-16:
@wkentaro Thank you. It does indeed seems to be fixed by 3ad75a15430ce46b25341233665bcce40e3afa70 .

Haini, 2016-10-06:
I hate to reopen this (especially as I don't know if this is the right place) - but I am running in pretty much the same error on Debian 8.0 Sid while using the [Debian Robotics Repository](https://wiki.debian.org/DebianScience/Robotics/ROS#CA-82922f5318527584dec613afe4df73eec3c6fd34_4).

Everything builds fine until we come to the image_view part:

> ==> make -j4 -l4 in '/home/consti/ros/build_isolated/image_view'
> [  5%] Generating dynamic reconfigure files from cfg/ImageView.cfg: /home/consti/ros/devel_isolated/image_view/include/image_view/ImageViewConfig.h /home/consti/ros/devel_isolated/image_view/lib/python2.7/dist-packages/image_view/cfg/ImageViewConfig.py
> [ 29%] Built target video_recorder
> [ 29%] Built target extract_images
> [ 41%] Built target stereo_view
> [ 52%] Built target disparity_view
> [ 64%] Built target image_saver
> [ 88%] Built target image_view
> Generating reconfiguration files for ImageView in image_view
> [ 88%] Built target image_view_gencfg
> [ 94%] Building CXX object CMakeFiles/image_view_exe.dir/src/nodes/image_view.cpp.o
> /home/consti/ros/src/image_view/src/nodes/image_view.cpp:34:40: **fatal error: image_view/ImageViewConfig.h: No such file or directory**
>  #include <image_view/ImageViewConfig.h>
>                                         ^
> compilation terminated.
> CMakeFiles/image_view_exe.dir/build.make:62: recipe for target 'CMakeFiles/image_view_exe.dir/src/nodes/image_view.cpp.o' failed
> make[2]: *** [CMakeFiles/image_view_exe.dir/src/nodes/image_view.cpp.o] Error 1
> CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/image_view_exe.dir/all' failed
> make[1]: *** [CMakeFiles/image_view_exe.dir/all] Error 2
> Makefile:138: recipe for target 'all' failed
> make: *** [all] Error 2
> <== Failed to process package 'image_view': 
>   Command '['/home/consti/ros/install_isolated/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2
> 
> Reproduce this error by running:
> ==> cd /home/consti/ros/build_isolated/image_view && /home/consti/ros/install_isolated/env.sh make -j4 -l4

Is there any probability that the issue is on the side of the debian repo?

vrabaud, 2016-10-10:
It builds fine on Jessie: http://build.ros.org/view/Ksrc_dJ/job/Kbin_dj_dJ64__ecto_image_pipeline__debian_jessie_amd64__binary/12/consoleFull
The header gets built before so it's really that your instruction is not aware of the include folder.
Which version are you using ? Which ROS distro ?

Haini, 2016-10-10:
I am 3 days wiser on this issue now, realizing that the build system changed over time (e.g catkin_make, catkin build, rosbuild) and I mixed some of these up massively as desperation grew bigger.

The install I attempted here was from ROS Indigo on Debian Jessie Unstable I think. After lots of missing dependencies I started a new try with ROS Kinetic on a clean Debian install and that one worked out fine. 

I guess all the broken Makefiles were my fault for mixing up buildtools and dependency resolvers. 

Thanks for your reply and time!

PR #186 (2016-03-13): [image_view] Fix #185: not found ImageViewConfig.h
Fix #185
commit 3ad75a15 (2016-03-13): Remove code for roslib on .cfg files

Closes #185
commit 53c87cd1 (2016-03-13): Merge pull request #186 from wkentaro/image-view-config

[image_view] Fix #185: not found ImageViewConfig.h