ros-perception/image_pipeline #187: image_view video video_recorder not work

randoms, 2016-03-16:
After running

    rosrun image_view video_recorder image:=/camera_node/image_raw

I got the following error

    [ERROR] [1458111249.024500050]: Unable to convert rgb8 image to rgb8

I edited the source code to print the exception

    cv_bridge.cvtColorForDisplay() scaling for images with more than one channel is unsupported

So the real problem is not video format.

In  [video_recorder.cpp Line 175](https://github.com/ros-perception/image_pipeline/blob/indigo/image_view/src/nodes/video_recorder.cpp#L75), use_dynamic_range was set to false, and min_depth_range=0.0, max_depth_range=5.5. According to the [doc](http://docs.ros.org/jade/api/cv_bridge/html/c++/namespacecv__bridge.html#a4a07b9a28883c6f97a6449a1bc69a79f), "Independently from do_dynamic_scaling, if min_image_value and max_image_value are different, the image is scaled between these two values before being converted to its final encoding. "

I think you can either print a better error message or set min_image_value and max_image_value to be 0.

anuppari, 2016-03-25:
I have had a similar issue. When trying to record a mono8 video, the saved file is all blown out. When I change the `max_depth_range` back to 0.0, it works fine. I think the default value that `cv_bridge::cvtColorForDisplay` uses (i.e., 0.0) is correct since that doesn't change anything without telling the user, but the node uses a default value of 5.5. The min/max depth range and dynamic depth range parameters are not even listed on the ROS wiki page for this node, so there is no indication to the user why the video recording is bad despite `image_view` showing the correct image.

ferdianjovan, 2016-04-05:
I got the same problem. Thanks there is a solution for this.

andrea-nisti, 2016-04-06:
hello everyone, i had the same error but i solved it putting max and min range to 0. However, when i look at output.avi i can't see the video on vlc. Did someone experienced something like that?

wkentaro, 2016-06-14:
Same problem here.

closed 2016-07-24 by vrabaud

vrabaud, 2016-07-24:
new version released in shadow-fixed.