ros-perception/image_pipeline #188: stereo_view showing empty left, right, disparity windows with opencv3

mtbrobotanist, 2016-03-21:
hello, myself and at least the poster of the question at this url: 

http://answers.ros.org/question/225012/image_view-stereo_view-do-not-show-image-or-disparity/?answer=229690#post-id-229690

have experienced stereo view showing empty windows for the left, right and disparity images. 

I  got it to work again by changing some code in the image call back function. My answer to the question at the above url details the solution. 

I was also asked to post to the issue here by the moderator but I've never done this before so i don't know how much more detail is needed sorry!

jack-oquin, 2016-03-21:
If possible, describe how to demonstrate the bug reliably. Since this looks timing-related, that might not be possible, but "what you were doing when you saw the error" would still help. Otherwise, how can we tell that a fix actually works?

Your sketch at a solution in the answer linked above is quite helpful. If you wish to assist the maintainers even more, you can:

* make a fork of this repository
* apply the fix to your copy
* test that it does indeed fix the bug for you
* submit a github pull request referencing this issue (as #188)

godo2, 2016-06-06:
I think I have the same problem; The Image pipeline works with the binary version in ROS (Jade-desktop-full, Ubuntu 14.04) very well. But when I use the source code from github in workspace with  catkin_make or (catkin_make -DCMAKE_INSTALL_PREFIX=/opt/ros/jade install), then delivers the stereo process an error. The error appears after calling the stereo_view nodelets with empty pictures and an error code.

The change in the code image_view / stereo_view.cpp brought no improvements.

Does anyone have any idea why this might be

Error code:
http://answers.ros.org/question/235342/image_pipeline/

godo2, 2016-06-08:
Correction: the described problem appears at a opencv version 2.x and is to correct by modifying the code in stereo_view.cpp.

cwecht, 2016-12-20:
The problem can be reproduced by doing [this tutorial](http://wiki.ros.org/stereo_image_proc/Tutorials/ChoosingGoodStereoParameters).

PR #242 (2017-01-02): stereo_view: fixed empty left, right, disparity windows with opencv3
Fixed #188 (the solution is described there).
closed 2017-01-02 by vrabaud