ros-perception/image_pipeline #191: add ArUco and ChArUco board calibration as an option

ezavesky, 2016-04-15:
OpenCV3 has introduced a free-to-use powerful AR marker [called ArUco](http://www.uco.es/investiga/grupos/ava/node/26). Also introduced in this code set is the ability to calculate a pose or calibration set [along with a tutorial](http://docs.opencv.org/3.1.0/df/d4a/tutorial_charuco_detection.html).  The current board calibration works well enough, but in future generations (like when CV3 is more standard), there might be large community benefit to get this working properly.

If the time comes up, I might consider it in the future, this issue might tease out other volunteers in the mean time.

(some inline links in case the opencv or author page go away in the future)
`detect_board_charuco.cpp` inside the module samples folder of opencv 3

[image]

vrabaud, 2016-04-15:
+1
Well, OpenCV3 is the default in Kinetic so have fun :)

ezavesky, 2016-04-15:
That's great news.  We're using one of those environments still based on `indigo`; do you know if that branch's version of these modules play as well with OCV3 or would some back-pulls be required?

vrabaud, 2016-04-15:
It's easy to write code that works with OpenCV2 and 3, cf http://wiki.ros.org/opencv3 but I won't have time for that. OpenCV3 is packaged in Indigo but not the default, so you will have to recompile your whole ROS stacks to use OpenCV3.

JStech, 2020-06-11:
This issue looks stale, but I'd like to have ChArUco board support. I'll see if I can get it working.

SteveMacenski, 2020-06-11:
@JStech sounds good! Its mostly stale because no one did it. If it works and it was done relatively cleanly, no issue merging it.

ezavesky, 2020-06-16:
Thanks folks, sorry that I'm not in that project space to help any more.  That said it still looks like a slick feature for calibration. +1 @JStech

JWhitleyWork, 2020-07-29:
Looks like @JStech took care of adding this feature to melodic and noetic. Thanks for that! We'll handle updating ROS2 branches with this feature as part of #449.

closed 2020-07-29 by JWhitleyWork