ros-perception/image_pipeline #193: catkin_make error   Could not find a package configuration file provided by "tf2_geometry_msgs"

Tondy, 2016-05-02:
Hello I git cloned the package in the src directory and receive the error below when trying to make the package:

CMake Warning at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "tf2_geometry_msgs"
  with any of the following names:

    tf2_geometry_msgsConfig.cmake
    tf2_geometry_msgs-config.cmake

  Add the installation prefix of "tf2_geometry_msgs" to CMAKE_PREFIX_PATH or
  set "tf2_geometry_msgs_DIR" to a directory containing one of the above
  files.  If "tf2_geometry_msgs" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  image_pipeline/image_rotate/CMakeLists.txt:4 (find_package)

-- Could not find the required component 'tf2_geometry_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "tf2_geometry_msgs"
  with any of the following names:

    tf2_geometry_msgsConfig.cmake
    tf2_geometry_msgs-config.cmake

  Add the installation prefix of "tf2_geometry_msgs" to CMAKE_PREFIX_PATH or
  set "tf2_geometry_msgs_DIR" to a directory containing one of the above
  files.  If "tf2_geometry_msgs" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  image_pipeline/image_rotate/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/odroid/drone/build/CMakeFiles/CMakeOutput.log".
See also "/home/odroid/drone/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

vrabaud, 2016-05-02:
image_rotate does include tf2_geometry_msgs in its rosdep as shown at: https://github.com/ros-perception/image_pipeline/blob/indigo/image_rotate/package.xml#L42
That package also builds fine on the farm.
Is tf2_geoemetry_msgs installed ? What is specific about your platform ? (I see you have an odroid),

Tondy, 2016-05-02:
Hello vrabaud

The specifies are I am running ubuntu 14 on a odroid ux4.

Tondy, 2016-05-02:
The solution is some where here:

Add the installation prefix of "tf2_geometry_msgs" to CMAKE_PREFIX_PATH or
set "tf2_geometry_msgs_DIR" to a directory containing one of the above
files. If "tf2_geometry_msgs" provides a separate development package or
SDK, be sure it has been installed.
Call Stack (most recent call first):

But do not understand what I have to actually do.

vrabaud, 2016-05-02:
Do you have it installed ? Can you do `roscd tf2_geometry_msgs` ?

Tondy, 2016-05-02:
roscd: No such package/stack 'tf2_geometry_msgs'

When I git clone tf2_geometry_msgs and try and catkin_make:

-- Using these message generators: gencpp;genlisp;genpy
-- Configuring incomplete, errors occurred!
See also "/home/odroid/drone/build/CMakeFiles/CMakeOutput.log".
See also "/home/odroid/drone/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

Is they a way to sudo apt-get install ?

vrabaud, 2016-05-03:
I don't know your setup. Please follow instructions at http://wiki.ros.org/jade/Installation/Source or look at similar errors on http://answers.ros.org/

closed 2016-07-24 by vrabaud

Instassa, 2017-08-25:
I have exactly the same problem/error - it's a different package, but everything else is the same:
     moveit_visual_toolsConfig.cmake
     moveit_visual_tools-config.cmake
Has this ever been fixed?

SabyasachiDash, 2018-05-18:
use -- $ sudo apt-get install ros-kinetic-tf2-geometry-msgs

arjunskumar, 2021-04-05:
@SabyasachiDash This worked

sifaqes, 2021-07-19:
> use -- $ sudo apt-get install ros-kinetic-tf2-geometry-msgs

sudo apt-get install ros-kinetic-tf2-geometry-msgs 

this is  not worked for me  . i have this error when i will install pakage ROSSERIAL . anyone can help me  

-- +++ processing catkin package: 'rosserial_windows'
-- ==> add_subdirectory(rosserial/rosserial_windows)
-- Could NOT find geometry_msgs (missing: geometry_msgs_DIR)
-- Could not find the required component 'geometry_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /home/pi/ros_catkin_ws/devel/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "geometry_msgs"
  with any of the following names:

    geometry_msgsConfig.cmake
    geometry_msgs-config.cmake

  Add the installation prefix of "geometry_msgs" to CMAKE_PREFIX_PATH or set
  "geometry_msgs_DIR" to a directory containing one of the above files.  If
  "geometry_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  rosserial/rosserial_windows/CMakeLists.txt:4 (find_package)

arjunskumar, 2021-07-19:
@sifaqes Please try `sudo apt-get install ros-kinetic-geometry-msgs`

sifaqes, 2021-07-19:
No worked, i have raspberry pi 5

arjunskumar, 2021-07-19:
@sifaqes Which version of ROS are using?

sifaqes, 2021-07-19:
pakage ros, desktop (recomanded ) en raspberry pi 4 
 * /rosdistro: kinetic
 * /rosversion: 1.12.17

I have installed common_msgs and I solve the problem, but I have a new fault with the resserial library, these are the packages

https://github.com/ros-drivers/rosserial.git

this is  new probleme  

make[2]: *** [rosserial/rosserial_server/CMakeFiles/rosserial_server_serial_node.dir/build.make:63: rosserial/rosserial_server/CMakeFiles/rosserial_server_serial_node.dir/src/serial_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:20225: rosserial/rosserial_server/CMakeFiles/rosserial_server_serial_node.dir/all] Error 2
make[1]: *** Se espera a que terminen otras tareas....
make[2]: *** [rosserial/rosserial_server/CMakeFiles/rosserial_server_udp_socket_node.dir/build.make:63: rosserial/rosserial_server/CMakeFiles/rosserial_server_udp_socket_node.dir/src/udp_socket_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:20313: rosserial/rosserial_server/CMakeFiles/rosserial_server_udp_socket_node.dir/all] Error 2
make[2]: *** [rosserial/rosserial_server/CMakeFiles/rosserial_server_socket_node.dir/build.make:63: rosserial/rosserial_server/CMakeFiles/rosserial_server_socket_node.dir/src/socket_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:20374: rosserial/rosserial_server/CMakeFiles/rosserial_server_socket_node.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j4 -l4" failed
pi@raspberrypi:~/ros_catkin_ws$

arjunskumar, 2021-07-19:
@sifaqes

 ```
cd ros_catkin_ws/src/
 git clone https://github.com/ros/common_msgs 
cd .. && catkin_make 
```

This will fix your problem

sifaqes, 2021-07-19:
Its fixed now, thanks,  for pakage rosserial i am swaping to rosserial kinetic and was work

sifaqes, 2021-07-19:
@arjunskumar 
i have  quastion whats the defirent betwen catkin_make  and  sudo ./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release --install-space /opt/ros/kinetic ?

sahilsaqi, 2024-02-23:
If someone is using ROS noetic raspi -4 
and get same error try this 

cd ros_catkin_ws/src/

git clone https://github.com/ros/geometry2.git

cd ros_catkin_ws
catkin_make