ros-perception/image_pipeline #196: depth_image_proc nodelets crash on load on debian Jessie in Kinetic

tfoote, 2016-05-09:
I've reproduced this on amd64 aarch64 in hardware and it is reproduceable in docker too. 

Using the Jessie debs in shadow-fixed:
```
$ sudo apt-get Install ros-kinetic-depth-image-proc
$ roscore&
$ rosrun nodelet nodelet standalone depth_image_proc/point_cloud_xyzrgb
type is depth_image_proc/point_cloud_xyzrgb
Segmentation fault
```

Here's the backtrace in gdb:
```
(gdb) r standalone depth_image_proc/point_cloud_xyz
Starting program: /opt/ros/kinetic/lib/nodelet/nodelet standalone depth_image_proc/point_cloud_xyz
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
type is depth_image_proc/point_cloud_xyz
[New Thread 0x7fffefa9b700 (LWP 11688)]
[New Thread 0x7fffef29a700 (LWP 11689)]
[New Thread 0x7fffeea99700 (LWP 11690)]
[New Thread 0x7fffee298700 (LWP 11695)]
[New Thread 0x7fffed46c700 (LWP 11697)]
[New Thread 0x7fffecc6b700 (LWP 11698)]
[New Thread 0x7fffdffff700 (LWP 11699)]
[New Thread 0x7fffdf7fe700 (LWP 11700)]
[New Thread 0x7fffdeffd700 (LWP 11701)]

Program received signal SIGSEGV, Segmentation fault.
0x00007fffc86cfcdc in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
(gdb) bt full
#0  0x00007fffc86cfcdc in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
No symbol table info available.
#1  0x00007ffff7dea9ba in call_init (l=<optimized out>, argc=argc@entry=3, 
    argv=argv@entry=0x7fffffffd818, env=env@entry=0x7fffffffd838) at dl-init.c:78
        j = <optimized out>
        jm = <optimized out>
        addrs = <optimized out>
        init_array = <optimized out>
#2  0x00007ffff7deaaa3 in call_init (env=0x7fffffffd838, argv=0x7fffffffd818, argc=3, l=<optimized out>)
    at dl-init.c:36
No locals.
#3  _dl_init (main_map=main_map@entry=0xa40a60, argc=3, argv=0x7fffffffd818, env=0x7fffffffd838)
    at dl-init.c:126
        preinit_array = <optimized out>
        preinit_array_size = <optimized out>
        i = 119
#4  0x00007ffff7deec08 in dl_open_worker (a=a@entry=0x7fffffffc408) at dl-open.c:577
        args = 0x7fffffffc408
        file = <optimized out>
        mode = <optimized out>
        call_map = <optimized out>
        dst = <optimized out>
        __PRETTY_FUNCTION__ = "dl_open_worker"
        new = <optimized out>
        r = <optimized out>
        reloc_mode = <optimized out>
        nmaps = <optimized out>
        l = <optimized out>
        maps = <optimized out>
        any_tls = <optimized out>
        first_static_tls = <optimized out>
#5  0x00007ffff7dea874 in _dl_catch_error (objname=objname@entry=0x7fffffffc3f8, 
    errstring=errstring@entry=0x7fffffffc400, mallocedp=mallocedp@entry=0x7fffffffc3f7, 
    operate=operate@entry=0x7ffff7dee930 <dl_open_worker>, args=args@entry=0x7fffffffc408)
    at dl-error.c:187
        errcode = 0
        c = {objname = 0x7fffffffc3f8, errstring = 0x7fffffffc400, malloced = 0x7fffffffc3f7, 
          errcode = 0x7fffffffc2e4, env = {{__jmpbuf = {140737488340232, 3137333368337985864, 
                2147483905, 10808776, 140737488345112, 3, 3137333368535118152, 3137351226111422792}, 
              __mask_was_saved = -136426679, __saved_mask = {__val = {140737488340096, 2762155452, 24, 
                  45, 140737353865328, 140737351931278, 140737488340112, 140737488339904, 
                  140737293385488, 140737293392832, 140737488340176, 43158678, 140737488340160, 
                  140737334932392, 0, 0}}}}}
        catchp = 0x7ffff7fb7708
        old = <optimized out>
#6  0x00007ffff7dee3fb in _dl_open (file=0xa4edc8 "/opt/ros/kinetic/lib//libdepth_image_proc.so", 
    mode=-2147483391, caller_dlopen=<optimized out>, nsid=-2, argc=3, argv=0x7fffffffd818, 
    env=0x7fffffffd838) at dl-open.c:661
        args = {file = 0xa4edc8 "/opt/ros/kinetic/lib//libdepth_image_proc.so", mode = -2147483391, 
          caller_dlopen = 0x7ffff707e212 <Poco::SharedLibraryImpl::loadImpl(std::string const&)+66>, 
          caller_dl_open = 0x7ffff6db002b <dlopen_doit+91>, map = 0xa40a60, nsid = 0, argc = 3, 
---Type <return> to continue, or q <return> to quit---
          argv = 0x7fffffffd818, env = 0x7fffffffd838}
        objname = 0x100000132 <error: Cannot access memory at address 0x100000132>
        errstring = 0x1 <error: Cannot access memory at address 0x1>
        malloced = false
        errcode = <optimized out>
        __PRETTY_FUNCTION__ = "_dl_open"
#7  0x00007ffff6db002b in dlopen_doit (a=a@entry=0x7fffffffc620) at dlopen.c:66
        args = 0x7fffffffc620
#8  0x00007ffff7dea874 in _dl_catch_error (objname=0x636830, errstring=0x636838, mallocedp=0x636828, 
    operate=0x7ffff6daffd0 <dlopen_doit>, args=0x7fffffffc620) at dl-error.c:187
        errcode = 0
        c = {objname = 0x636830, errstring = 0x636838, malloced = 0x636828, errcode = 0x7fffffffc504, 
          env = {{__jmpbuf = {0, 3137353462168416584, 140737488340512, 0, 6511264, 9654400, 
                3137333368329597256, 3137351226111422792}, __mask_was_saved = -136405561, 
              __saved_mask = {__val = {140733193388037, 0, 0, 140737293392832, 6514720, 
                  140737334935504, 140737488340512, 140737351975653, 6514720, 5, 40, 6514720, 3, 
                  6514720, 140737334935504, 140737488340512}}}}}
        catchp = 0x7ffff7fb7708
        old = <optimized out>
#9  0x00007ffff6db05dd in _dlerror_run (operate=operate@entry=0x7ffff6daffd0 <dlopen_doit>, 
    args=args@entry=0x7fffffffc620) at dlerror.c:163
        result = 0x636820
#10 0x00007ffff6db00c1 in __dlopen (file=<optimized out>, mode=<optimized out>) at dlopen.c:87
        args = {file = 0xa4edc8 "/opt/ros/kinetic/lib//libdepth_image_proc.so", mode = 257, 
          new = 0x635a50, 
          caller = 0x7ffff707e212 <Poco::SharedLibraryImpl::loadImpl(std::string const&)+66>}
#11 0x00007ffff707e212 in Poco::SharedLibraryImpl::loadImpl(std::string const&) ()
   from /usr/lib/libPocoFoundation.so.9
No symbol table info available.
#12 0x00007ffff707e800 in Poco::SharedLibrary::SharedLibrary(std::string const&) ()
   from /usr/lib/libPocoFoundation.so.9
No symbol table info available.
#13 0x00007ffff731d7cb in class_loader::class_loader_private::loadLibrary(std::string const&, class_loader::ClassLoader*) () from /opt/ros/kinetic/lib/libclass_loader.so
No symbol table info available.
#14 0x00007ffff73173c0 in class_loader::ClassLoader::loadLibrary() ()
   from /opt/ros/kinetic/lib/libclass_loader.so
No symbol table info available.
#15 0x00007ffff7317618 in class_loader::ClassLoader::ClassLoader(std::string const&, bool) ()
   from /opt/ros/kinetic/lib/libclass_loader.so
No symbol table info available.
#16 0x00007ffff73214d8 in class_loader::MultiLibraryClassLoader::loadLibrary(std::string const&) ()
   from /opt/ros/kinetic/lib/libclass_loader.so
No symbol table info available.
#17 0x00007ffff7babedc in pluginlib::ClassLoader<nodelet::Nodelet>::loadLibraryForClass(std::string const&) () from /opt/ros/kinetic/lib/libnodeletlib.so
No symbol table info available.
#18 0x00007ffff7bb6e6c in pluginlib::ClassLoader<nodelet::Nodelet>::createInstance(std::string const&)
    () from /opt/ros/kinetic/lib/libnodeletlib.so
No symbol table info available.
#19 0x00007ffff7ba54f4 in boost::detail::function::function_obj_invoker1<boost::_bi::bind_t<boost::shared---Type <return> to continue, or q <return> to quit---
_ptr<nodelet::Nodelet>, boost::_mfi::mf1<boost::shared_ptr<nodelet::Nodelet>, pluginlib::ClassLoader<nodelet::Nodelet>, std::string const&>, boost::_bi::list2<boost::_bi::value<boost::shared_ptr<pluginlib::ClassLoader<nodelet::Nodelet> > >, boost::arg<1> > >, boost::shared_ptr<nodelet::Nodelet>, std::string const&>::invoke(boost::detail::function::function_buffer&, std::string const&) ()
   from /opt/ros/kinetic/lib/libnodeletlib.so
No symbol table info available.
#20 0x00007ffff7ba3af4 in nodelet::Loader::load(std::string const&, std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&, std::vector<std::string, std::allocator<std::string> > const&) ()
   from /opt/ros/kinetic/lib/libnodeletlib.so
No symbol table info available.
#21 0x0000000000404efc in main ()
No symbol table info available.
```

@j-rivero this is what I was debugging. After testing I've found it's not isolated to arm64. 
@mikaelarguedas FYI re nodelets  I've verified that other nodelets seem to work fine. At least the tutorial nodelets run, there are not a lot to test with yet in kinetic/Jessie.

vrabaud, 2016-05-09:
Well, first things first: why does it end up in Qt4 ?

Because OpenCV3 depends on Qt4 on Jessie: http://build.ros.org/view/Kbin_dJ64/job/Kbin_dJ64__opencv3__debian_jessie_amd64__binary/17/consoleFull
Why ? Because OpenCV3 depends on VTK and VTK5 got chosen on Jessie: https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L2683
during commit: https://github.com/ros/rosdistro/commit/88c5d1ed3ceb16bb16aa91dc246a5a701907be5d

But VTK5 depends on libqt4-dev on Jessie: https://packages.debian.org/jessie/libvtk5-dev which is against REP 003. Soooooooooo, let's continue the conversation here: https://github.com/ros/rosdistro/pull/11240

vrabaud, 2016-05-09:
I can reproduce (yay, Jessie VM ...). And I can indeed confirm that the libdepth_image_proc.so is linked to both libQt5Gui.so.5 and libQtCore.so.4. So I won't investigate any further until the rosdep keys are fixed, that already seems like a good reason for a crash.

vrabaud, 2016-05-11:
Hi, fixing the dependencies of OpenCV3 fixed the trick on my Jessie VM. So we are good.

closed 2016-05-11 by vrabaud