ros-perception/image_pipeline #197: could not convert ‘use_dynamic_range’ from ‘bool’ to ‘cv_bridge::CvtColorForDisplayOptions’

Yayi, 2016-05-20:
image_view:

could not convert ‘use_dynamic_range’ from ‘bool’ to ‘cv_bridge::CvtColorForDisplayOptions’
 Display(cv_bridge::toCvShare(image_msg), encoding, use_dynamic_range, min_depth_range, max_depth_range)->image;

CC:GCC6

mikepurvis, 2016-06-01:
I'm seeing a similar error:

```
image_view/src/nodelets/image_nodelet.cpp:165:87: error: could not convert 'do_dynamic_scaling' from 'bool' to 'cv_bridge::CvtColorForDisplayOptions'
last_image_ = cvtColorForDisplay(cv_bridge::toCvShare(msg), "", do_dynamic_scaling)->image;
```

mikepurvis, 2016-06-01:
I think the issue is in cv_bridge, introduced here: https://github.com/ros-perception/vision_opencv/commit/2b75bdc13b3b41ed65d4d66e4eae16453e1d0242

wkentaro, 2016-06-01:
If you install the vision_opencv from source, [this PR](https://github.com/ros-perception/image_pipeline/pull/194) is required to compile image_pipeline.
cv_bridge is not released yet after the commit, and this issue is what is expected. https://github.com/ros-perception/vision_opencv/commit/2b75bdc13b3b41ed65d4d66e4eae16453e1d0242

mikepurvis, 2016-06-01:
Indigo is an LTS release, and `vision_opencv` and `image_pipeline` are core packages.

@tfoote can clarify, but IMO it would be best if this development occurred on a Kinetic branch, with the Indigo interfaces considered stable.

wkentaro, 2016-06-01:
Sounds reasonable, but I only have a read right for these repos, and cannot decide about that.
@vrabaud What do you think about this? Could you please create `jade` or `kinetic` branch?

vrabaud, 2016-06-14:
terribly sorry to be so late on that one. yes, it was fault btw. Solutions:
 - leave it as is and release cv_bridge asap on Indigo/Kinetic: this would be my favortie solution as I don't have time to maintain several branches for image_pipeline. This API is only for GUIs (and only image_view right now) so I don't think this symbol breakage would break any robot
- revert on Indigo cv_bridge only but add a define there (and not on Kinetic) that would help maintain one branch only for image_pipeline.
- revert on Indigo/Kinetic but we would lose a great feature that should end up in rqt_image_view at some point.

So, can I go with one pretty please ? :)

wkentaro, 2016-06-15:
I agree with the first solution.

mikepurvis, 2016-06-15:
I don't care as much about ABI issues, but I'd like the devel branches to compile and work at any given time.

wkentaro, 2016-07-08:
@pete-bot Maybe not, because https://github.com/ros-perception/image_pipeline/pull/194 is not merged yet.

wkentaro, 2016-07-08:
Maybe you can use version `1.12.0` of `vision_opencv`.

vrabaud, 2016-07-22:
Fixed with the latest version of vision_opencv (in shadow-fixed for now). That bug lasted way too long, really sorry about that.

closed 2016-07-22 by vrabaud