ros-perception/image_pipeline #202: Image_view with nodelet

godo2, 2016-06-16:
Is it possible in the package image_view to use the nodelet (for disparity)? 
I've tried it with the following launch file. But it appeared only a blank window.

  <node pkg="nodelet" type="nodelet" name="disparity_view"
        args="load image_view/disparity stereo:=/stereo_1 image:=disparity $(arg manager)" />

wkentaro, 2016-12-03:
It should be something like below:
```xml
<node pkg="nodelet" type="nodelet" name="disparity_view"
      args="load image_view/disparity $(arg manager)">
  <remap from="stereo" to="/stereo_1" />
  <remap from="image" to="disparity" />
</node>
```

vrabaud, 2016-12-04:
Please ask your questions on http://answers.ros.org/questions/

closed 2016-12-04 by vrabaud