ros-perception/image_pipeline #204: stereo-image-proc dies upon subscription (building from src)

Karnix, 2016-06-24:
Launch file being used:
`<launch>
  <node ns="stereo" pkg="uvc_camera" type="uvc_stereo_node" name="uvc_camera_stereo">
    <param name="width" type="int" value="320" />
    <param name="height" type="int" value="240" />
    <param name="fps" type="int" value="30" />
    <param name="frame" type="string" value="wide_stereo" />

    <param name="auto_focus" type="bool" value="False" />
    <param name="focus_absolute" type="int" value="0" />
    
    

    <param name="left/device" type="string" value="/dev/video1" />
    <param name="right/device" type="string" value="/dev/video2" />
    <param name="left/camera_info_url" type="string" value="file:///home/karnix/ROS_Work/src/logitech/left.yaml" />
    <param name="right/camera_info_url" type="string" value="file:///home/karnix/ROS_Work/src/logitech/right.yaml" />
  </node>

 <node ns="stereo" pkg="stereo_image_proc" type="stereo_image_proc" name="stereo_image_proc" >
    <param name="approximate_sync" type="bool" value="True" />
</node>

  </launch>
`
Upon launching this file, the stereo_image_proc and uvc_camera_stereo nodes are running.

But when I try to subscribe to the /stereo/disparity, /stereo/right/image_rect and /stereo/left/image_rect by running the following command,

`rosrun image_view stereo_view stereo:=stereo image:=rect_image_color`

the process dies with the following message

`[stereo/stereo_image_proc-3] process has died [pid 12227, exit code -11, cmd /home/karnix/ros_stereo/devel/lib/stereo_image_proc/stereo_image_proc __name:=stereo_image_proc __log:=/home/karnix/.ros/log/03a74da8-39cf-11e6-bc56-28b2bdcaf7d8/stereo-stereo_image_proc-3.log].
log file: /home/karnix/.ros/log/03a74da8-39cf-11e6-bc56-28b2bdcaf7d8/stereo-stereo_image_proc-3*.log
`
And I found the log file to be empty.

Any solutions?

godo2, 2016-06-24:
look here, may help

https://github.com/ros-perception/image_pipeline/issues/188

http://answers.ros.org/question/235342/image_pipeline/

Karnix, 2016-06-24:
No, they don't help :/

SteveMacenski, 2019-11-04:
This crash doesn't tell us anything without compiling with GDB, please reopen with a backtrace or ask on ROS Answers

closed 2019-11-04 by SteveMacenski