ros-perception/image_pipeline #209: [imgwarp.cpp] crashes: OpenCV Error: Assertion failed (_map1.size().area() > 0) in remap

willdzeng, 2016-07-09:
I have a node that publishing following topic under namespace "camera"

```
/left/image_raw
/left/camera_info
/right/image_raw
/right/camera_info
```

and I am trying to use **stereo_image_proc** node to rectify the image.

`<node pkg="stereo_image_proc" type="stereo_image_proc" name="stereo_image_proc" respawn="true" output="screen" ns="camera"/>`

but the node crashes when I subscribe the image_rect from rviz.

OpenCV Error: Assertion failed (_map1.size().area() > 0) in remap, file /home/zeng/essential/opencv/modules/imgproc/src/imgwarp.cpp, line 4833
terminate called after throwing an instance of 'cv::Exception'
  what():  /home/zeng/essential/opencv/modules/imgproc/src/imgwarp.cpp:4833: error: (-215) _map1.size().area() > 0 in function remap

I am using ROS Indigo, Ubuntu 14.04, OpenCV3.1
cv_bridge is 1.12.0 indigo branch build from source.
Image_pipeline is up-to-date indigo branch.

willdzeng, 2016-07-19:
Fixed by re-calibrate camera

closed 2016-07-19 by willdzeng

TZVIh, 2019-05-08:
Hi,
I have the same problem, I am using https://github.com/willdzeng/zed_cpu_ros
what should I do?
Thanks!

willdzeng, 2019-05-09:
As I mentioned previously, recalibrate would fix this problem, I suggest you directly use the opencv calibration package instead of using the build-in ZED calibration