ros-perception/image_pipeline #220: image_publisher_nodelet.cpp ambiguous overload error with boost 1.61/gcc 6.1.1

nxdefiant, 2016-08-18:
With Debian unstable I can not compile image_publisher for indigo with the following error:

```
/opt/ros/ros_catkin_ws/src/image_pipeline/image_publisher/src/nodelet/image_publisher_nodelet.cpp:180:43: error: ambiguous overload for 'operator=' (operand types are 'sensor_msgs::CameraInfo_<std::allocator<void> >::_D_type {aka std::vector<double>}' and 'boost::assign_detail::generic_list<int>')

camera_info_.D = list_of(0)(0)(0)(0)(0);
```

Changing the line to
`
camera_info_.D = {0, 0, 0, 0, 0};
`
seem to work.

nylocx, 2017-03-20:
Hi, thanks for sharing, I can confirm that this solves the error for me too on kinetic with GCC 6.3.1.

SteveMacenski, 2019-11-04:
Indigo is now EOL & fixed in Kinetic

closed 2019-11-04 by SteveMacenski