ros-perception/image_pipeline #222: [image_view] Unable to use image_view

yuki-asano, 2016-09-28:
I have a problem shown below in image_view.
Does anyone know the cause of this?

```
leus@leus-ThinkPad-T440p:/opt/ros/indigo/lib/image_view$ rosrun image_view image_view image:=/camera/rgb/image_rect_color
[ INFO] [1474958945.226256190]: Using transport "raw"
/opt/ros/indigo/lib/image_view/image_view: symbol lookup error: /opt/ros/indigo/lib/image_view/image_view: undefined symbol: _ZN9cv_bridge18cvtColorForDisplayERKN5boost10shared_ptrIKNS_7CvImageEEERKSsNS_25CvtColorForDisplayOptionsE
```

On the other hand, image_view2 works.
```
rosrun image_view2 image_view2 image:=/camera/rgb/image_rect_color
```

k-okada, 2016-09-28:
please run 
```
sudo apt-get update
sudo apt-get dist-upgrade
```
then try again.

yuki-asano, 2016-09-28:
It worked!
Thank you.

closed 2016-09-28 by yuki-asano