ros-perception/image_pipeline #228: rosrun camera_calibration cameracalibrator.py crashes on OSX

kubark42, 2016-10-06:
I can't quite tell from the crash dump if this is a python problem or a camera_calibration problem.

Command line:

```
ros/catkin_ws$ rosrun camera_calibration cameracalibrator.py image:=/camera/image_raw
Segmentation fault: 11
```

Crash dump:
```
Process:               Python [45985]
Path:                  /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            Python
Version:               2.7.10 (2.7.10)
Code Type:             X86-64 (Native)
Parent Process:        bash [506]
Responsible:           Python [45985]
User ID:               501

Date/Time:             2016-10-06 15:26:43.904 -0400
OS Version:            Mac OS X 10.12 (16A323)
Report Version:        12
Anonymous UUID:        6FBA6537-FAE4-1F9C-2B08-137FBC804364

Sleep/Wake UUID:       92626586-625F-4B13-8F85-0AD8A201C993

Time Awake Since Boot: 28000 seconds
Time Since Wake:       1000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

VM Regions Near 0:
--> 
    __TEXT                 0000000101079000-000000010107a000 [    4K] r-x/rwx SM=COW  /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   ???                           	000000000000000000 0 + 0
1   org.python.python             	0x00000001038b8995 PyEval_GetGlobals + 23
2   org.python.python             	0x00000001038c7e02 PyImport_Import + 137
3   org.python.python             	0x00000001038c6440 PyImport_ImportModule + 31
4   cv_bridge_boost.so            	0x00000001033c8e3a init_module_cv_bridge_boost() + 26
5   libboost_python-mt.dylib      	0x00000001033fb03e boost::python::handle_exception_impl(boost::function0<void>) + 78
6   libboost_python-mt.dylib      	0x00000001033fbfc9 boost::python::detail::init_module(char const*, void (*)()) + 121
7   org.python.python             	0x0000000101211190 _PyImport_LoadDynamicModule + 140
8   org.python.python             	0x000000010120fe8c 0x101176000 + 630412
9   org.python.python             	0x000000010120fa60 0x101176000 + 629344
10  org.python.python             	0x000000010120ec70 PyImport_ImportModuleLevel + 1185
11  org.python.python             	0x00000001011f4066 0x101176000 + 516198
12  org.python.python             	0x000000010118075b PyObject_Call + 99
13  org.python.python             	0x00000001011fee1b PyEval_CallObjectWithKeywords + 165
14  org.python.python             	0x00000001011fac6f PyEval_EvalFrameEx + 8283
15  org.python.python             	0x00000001011f8a1e PyEval_EvalCodeEx + 1617
16  org.python.python             	0x00000001011f83c7 PyEval_EvalCode + 48
17  org.python.python             	0x000000010120d71d PyImport_ExecCodeModuleEx + 241
18  org.python.python             	0x0000000101210427 0x101176000 + 631847
19  org.python.python             	0x00000001012106af 0x101176000 + 632495
20  org.python.python             	0x000000010120fe8c 0x101176000 + 630412
21  org.python.python             	0x000000010120faa1 0x101176000 + 629409
22  org.python.python             	0x000000010120ec38 PyImport_ImportModuleLevel + 1129
[... 380 lines cut ...]
TOTAL                            437.6M      780 

Model: MacBookPro11,5, BootROM MBP114.0172.B09, 4 processors, Intel Core i7, 2.5 GHz, 16 GB, SMC 2.30f2
Graphics: AMD Radeon R9 M370X, AMD Radeon R9 M370X, PCIe, 2048 MB
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x802C, 0x31364B544631473634485A2D314736453120
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x802C, 0x31364B544631473634485A2D314736453120
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x152), Broadcom BCM43xx 1.0 (7.21.171.10.1a16)
Bluetooth: Version 5.0.0f18, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0512G, 500.28 GB
USB Device: USB 3.0 Bus
USB Device: USB5434
USB Device: 4-Port USB 3.0 Hub
USB Device: AX88179
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Bluetooth USB Host Controller
USB Device: USB2134
USB Device: TUSB3410 Boot Device
USB Device: DELL USB Laser Mouse
USB Device: Composite Device
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1
```

spaghetti-, 2016-11-06:
is your `boost_python` upto date?

vrabaud, 2016-11-06:
@wjwwood , does that sound familiar ?

kubark42, 2016-11-06:
@spaghetti-, it wasn't, so I updated it to boost-python 1.62. Now everything ROS crashes (although I'm not sure the two are related because I didn't think to run ROS immediately before as a control).

```
roscore
... logging to /Users/kenz/.ros/log/f540394a-a437-11e6-a8fc-f45c89a27113/roslaunch-Kenns-MacBook-Pro.local-52818.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt

started roslaunch server http://Kenns-MacBook-Pro.local:61971/
ros_comm version 1.11.20

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.20

NODES

auto-starting new master
process[master]: started with pid [52825]
ROS_MASTER_URI=http://Kenns-MacBook-Pro.local:11311/

setting /run_id to f540394a-a437-11e6-a8fc-f45c89a27113
process[rosout-1]: started with pid [52828]
started core service [/rosout]
dyld: Symbol not found: __ZNK5boost16re_detail_10610031cpp_regex_traits_implementationIcE17transform_primaryEPKcS4_
  Referenced from: /opt/ros/indigo/lib/librosconsole.dylib
  Expected in: /usr/local/opt/boost/lib/libboost_regex-mt.dylib
```

Unfortunately, `catkin_make clean && catkin_make` didn't resolve the problem.

wjwwood, 2016-11-06:
I've not seen this before. If you update the version of boost Python you'll need to rebuild everything that used it. Also the boost Python is built against some specific version of Python, probably homebrews. Make sure ros is building against and using the same one.

SteveMacenski, 2019-09-03:
Closing from inactivity/reproducability -- if still an issue or still want to keep going lets revive it

closed 2019-09-03 by SteveMacenski