ros-perception/image_pipeline #235: Can't save image through image_view on Kinetic

wangji1, 2016-11-11:
On kinetic, when use mouse to right click image_view screen, it was failed to save the image.
And also image_view screen will be hang, even can't terminate it by press ctrl+c on terminal

k-okada, 2016-11-13:
is this same issue?? ->
https://github.com/ros-perception/image_pipeline/pull/184

wangji1, 2016-11-15:
similar...

Is that issue fixed?
but I have always upgraded ROS packages, so I think my used tool version should be latest, still has the problem.

If should I remove opencv, and try again?

k-okada, 2016-11-16:
which version of image_pipeline are you using, try rosversion image_view

wangji1, 2016-11-17:
$ rosversion image_view
1.12.19

wangji1, 2016-11-17:
[image]
This shows the image_view screen is hang

wangji1, 2016-11-17:
[image]
On terminal, there's no information for saved image

k-okada, 2016-11-17:
humm, it seems @iKrishneel 's fix has been merged at 1.12.16 so may be
different situation.
https://github.com/ros-perception/image_pipeline/blob/indigo/image_view/CHANGELOG.rst

iKrishneel, 2016-11-17:
@wangji1  
Did you build opencv from source?  That problems is likely result of OpenGL support. 

I fixed that by rebuilding OCV3 with this flag off: 
`-D WITH_OPENGL=OFF`

k-okada, 2016-11-17:
i think he's install them from deb, @Krishneel can you check if this
happens to your environment too?

wangji1, 2016-11-18:
Oh, I did download opencv source and build.

k-okada, 2016-11-18:
see #236 for the solution.

pbeeson, 2017-05-16:
I still see this behavior using the current kinetic .deb files.

k-okada, 2017-05-17:
may be that's because
https://github.com/ros-perception/image_pipeline/pull/236 is not merged and
released yet.

JWhitleyAStuff, 2019-04-22:
@pbeeson - I know this is a very stale issue but are you still seeing this behavior? @k-okada - Same question.

SteveMacenski, 2019-11-04:
Closing from non-responsive for 6 months. Feel free to re-open if still a problem in Melodic

closed 2019-11-04 by SteveMacenski