ros-perception/image_pipeline #243: image_proc failing to build on Arch

alphayed, 2016-12-27:
`[ 23%] Built target image_proc_gencfg
[ 84%] Built target image_proc
[ 92%] Linking CXX executable /home/alphayed/ROS/ros_catkin_ws/devel_isolated/image_proc/lib/image_proc/image_proc
/home/alphayed/ROS/ros_catkin_ws/install_isolated/lib/libopencv_viz3.so.3.1.0: undefined reference to `vtkSTLReader::New()'
/home/alphayed/ROS/ros_catkin_ws/install_isolated/lib/libopencv_viz3.so.3.1.0: undefined reference to `vtkOBJReader::New()'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/image_proc_exe.dir/build.make:219: /home/alphayed/ROS/ros_catkin_ws/devel_isolated/image_proc/lib/image_proc/image_proc] Error 1
make[1]: *** [CMakeFiles/Makefile2:68: CMakeFiles/image_proc_exe.dir/all] Error 2
make: *** [Makefile:139: all] Error 2
`
I've compile opencv3 from source but that did't help. Can you please guide me on this issue

vrabaud, 2017-01-02:
Did you compile your OpenCV3 or did you use the official ROS ones?

alphayed, 2017-01-02:
I installed the officially supported one using Arch's package manager. When image_proc failed while compiling ROS packages I removed it and used the one that is a part of ROS packages. Image_proc fails to compile either way.

vrabaud, 2017-01-03:
Hmmm, did you install VTK properly using rosdep ? Do you have the lib that contains vtkSTLReader::New() ?

alphayed, 2017-01-09:
I have found a fix as per this [thread](http://answers.ros.org/question/250846/compiling-ros-kinetec-from-source-fails-when-building-image_proc/). @vrabaud  Thank you for the help

closed 2017-01-09 by alphayed