ros-perception/image_pipeline #245: Calibration issue w fisheye lens

jmtatsch, 2017-01-18:
Hi,

we have some issues calibrating our U3-3241LE-C-HQ with a 2.5mm fisheye lens. Curiously our stronger 2.1mm fisheye lens works without issue. 

Here is the log:
 
rosrun camera_calibration cameracalibrator.py --size 8x6 --square 0.108 image:=/cam_front/image_raw camera:=/cam_front 

('Waiting for service', '/cam_front/set_camera_info', '...') OK *** Added sample 1, p_x = 0.525, p_y = 0.228, p_size = 0.260, skew = 0.160 *** Added sample 2, p_x = 0.515, p_y = 0.320, p_size = 0.279, skew = 0.057 *** Added sample 3, p_x = 0.405, p_y = 0.357, p_size = 0.324, skew = 0.004 *** Added sample 4, p_x = 0.314, p_y = 0.388, p_size = 0.316, skew = 0.155 *** Added sample 5, p_x = 0.214, p_y = 0.418, p_size = 0.288, skew = 0.232 *** Added sample 6, p_x = 0.129, p_y = 0.438, p_size = 0.268, skew = 0.144 *** Added sample 7, p_x = 0.124, p_y = 0.322, p_size = 0.240, skew = 0.075 *** Added sample 8, p_x = 0.232, p_y = 0.342, p_size = 0.257, skew = 0.001 *** Added sample 9, p_x = 0.518, p_y = 0.372, p_size = 0.310, skew = 0.197 *** Added sample 10, p_x = 0.585, p_y = 0.425, p_size = 0.339, skew = 0.295 *** Added sample 11, p_x = 0.721, p_y = 0.458, p_size = 0.343, skew = 0.209 *** Added sample 12, p_x = 0.812, p_y = 0.490, p_size = 0.336, skew = 0.138 *** Added sample 13, p_x = 0.761, p_y = 0.424, p_size = 0.275, skew = 0.274 *** Added sample 14, p_x = 0.745, p_y = 0.347, p_size = 0.291, skew = 0.392 *** Added sample 15, p_x = 0.816, p_y = 0.338, p_size = 0.269, skew = 0.491 *** Added sample 16, p_x = 0.884, p_y = 0.388, p_size = 0.242, skew = 0.553 *** Added sample 17, p_x = 0.889, p_y = 0.393, p_size = 0.246, skew = 0.313 *** Added sample 18, p_x = 0.955, p_y = 0.427, p_size = 0.224, skew = 0.462 *** Added sample 19, p_x = 0.844, p_y = 0.429, p_size = 0.212, skew = 0.213 *** Added sample 20, p_x = 0.428, p_y = 0.440, p_size = 0.304, skew = 0.112 *** Added sample 21, p_x = 0.452, p_y = 0.560, p_size = 0.275, skew = 0.073 *** Added sample 22, p_x = 0.442, p_y = 0.698, p_size = 0.262, skew = 0.115 *** Added sample 23, p_x = 0.420, p_y = 0.783, p_size = 0.289, skew = 0.026 *** Added sample 24, p_x = 0.412, p_y = 0.875, p_size = 0.295, skew = 0.128 *** Added sample 25, p_x = 0.340, p_y = 0.809, p_size = 0.314, skew = 0.191 *** Added sample 26, p_x = 0.264, p_y = 0.613, p_size = 0.314, skew = 0.017 *** Added sample 27, p_x = 0.250, p_y = 0.541, p_size = 0.327, skew = 0.158 *** Added sample 28, p_x = 0.140, p_y = 0.387, p_size = 0.302, skew = 0.452 *** Added sample 29, p_x = 0.222, p_y = 0.201, p_size = 0.272, skew = 0.393 *** Added sample 30, p_x = 0.296, p_y = 0.116, p_size = 0.259, skew = 0.338 *** Added sample 31, p_x = 0.489, p_y = 0.023, p_size = 0.326, skew = 0.277 *** Added sample 32, p_x = 0.615, p_y = 0.078, p_size = 0.316, skew = 0.213 *** Added sample 33, p_x = 0.602, p_y = 0.174, p_size = 0.297, skew = 0.293 *** Added sample 34, p_x = 0.501, p_y = 0.149, p_size = 0.264, skew = 0.340 *** Added sample 35, p_x = 0.428, p_y = 0.131, p_size = 0.237, skew = 0.235 *** Added sample 36, p_x = 0.250, p_y = 0.125, p_size = 0.229, skew = 0.517 *** Added sample 37, p_x = 0.061, p_y = 0.130, p_size = 0.204, skew = 0.781 *** Added sample 38, p_x = 0.096, p_y = 0.091, p_size = 0.221, skew = 0.666 *** Added sample 39, p_x = 0.660, p_y = 0.181, p_size = 0.318, skew = 0.160 *** Added sample 40, p_x = 0.740, p_y = 0.280, p_size = 0.310, skew = 0.140 *** Added sample 41, p_x = 0.630, p_y = 0.349, p_size = 0.326, skew = 0.109 *** Added sample 42, p_x = 0.457, p_y = 0.220, p_size = 0.305, skew = 0.026 *** Added sample 43, p_x = 0.394, p_y = 0.132, p_size = 0.370, skew = 0.150 *** Added sample 44, p_x = 0.457, p_y = 0.342, p_size = 0.453, skew = 0.424 *** Added sample 45, p_x = 0.476, p_y = 0.460, p_size = 0.423, skew = 0.507 *** Added sample 46, p_x = 0.361, p_y = 0.280, p_size = 0.466, skew = 0.378 *** Added sample 47, p_x = 0.381, p_y = 0.363, p_size = 0.457, skew = 0.271 *** Added sample 48, p_x = 0.357, p_y = 0.284, p_size = 0.506, skew = 0.195 *** Added sample 49, p_x = 0.267, p_y = 0.232, p_size = 0.430, skew = 0.253 *** Added sample 50, p_x = 0.158, p_y = 0.266, p_size = 0.410, skew = 0.319 *** Added sample 51, p_x = 0.446, p_y = 0.284, p_size = 0.419, skew = 0.567

*** Added sample 52, p_x = 0.490, p_y = 0.151, p_size = 0.445, skew = 0.544 *** Added sample 53, p_x = 0.448, p_y = 0.242, p_size = 0.415, skew = 0.324 *** Added sample 54, p_x = 0.556, p_y = 0.254, p_size = 0.432, skew = 0.198 *** Added sample 55, p_x = 0.679, p_y = 0.288, p_size = 0.440, skew = 0.073 ('D = ', [-0.7996277537451619, 0.520112755647464, -0.011141413854066822, 0.010632628154769112, 0.0]) ('K = ', [498.3820447204187, 0.0, 317.42096353490626, 0.0, 506.2957786823122, 261.8610192945964, 0.0, 0.0, 1.0]) ('R = ', [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]) ('P = ', [363.1578063964844, 0.0, 326.22065012135863, 0.0, 0.0, 374.8714599609375, 252.73824832158425, 0.0, 0.0, 0.0, 1.0, 0.0]) None # oST version 5.0 parameters   [image]  width 640  height 512  [narrow_stereo]  camera matrix 498.382045 0.000000 317.420964 0.000000 506.295779 261.861019 0.000000 0.000000 1.000000  distortion -0.799628 0.520113 -0.011141 0.010633 0.000000  rectification 1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000  projection 363.157806 0.000000 326.220650 0.000000 0.000000 374.871460 252.738248 0.000000 0.000000 0.000000 1.000000 0.000000

Interestingly the computed distortion parameters seem to be in the wrong direction:

[image]

Does anyone know what happens here and how to mitigate it?

garybradski, 2017-01-19:
Is this using the newer fisheye camera routines in OpenCV?
http://docs.opencv.org/3.1.0/d4/d94/tutorial_camera_calibration.html
http://docs.opencv.org/3.0-beta/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html

trigal, 2017-01-19:
uhm, wrong direction? could be related to #140 ? do you have this last version?

jmtatsch, 2017-01-23:
We are not yet using the newer opencv3 fisheye routines although we will definitely look into those in the future. We got a sort of working calibration now by using less skewed poses of the calibration board.

closed 2017-01-23 by jmtatsch