ros-perception/image_pipeline #246: [camera_calibration] Doesn't work well with mono16 images

mintar, 2017-01-23:
When running `cameracalibrator.py` on a `mono16` image stream, the image appears all white. This is because `cameracalibrator.py` simply crops all gray values to the lowest 256 values:

[calibrator.py:247](https://github.com/ros-perception/image_pipeline/blob/f94bfe7b03397a78cedd2686e8fb69d234b38447/camera_calibration/src/camera_calibration/calibrator.py#L247)

IMO, it should multiply each gray value by 1/256 instead.

I have verified this on my own image data: When I multiply the input image by 1/256 before feeding it into `cameracalibrator.py`, it works.

mintar, 2019-03-26:
There's an open pull request that solves this issue: #334

JWhitleyAStuff, 2019-04-25:
Closed by #334.

closed 2019-04-25 by JWhitleyAStuff