ros-perception/image_pipeline #251: How to use flip feature

bbidault2, 2017-02-09:
Hi all, first, thank you all for the contribution.
I know this issue has already been mentioned in #143 and was solved by #173, but I now have trouble to run the flip feature of image_publisher. rosrun tells me that I should run the node this way `rosrun image_publisher image_publisher <filename>`, how do I select the flip orientation (horizontal, vertical, both) ? Moreover, is it possible to run this feature on a topic ? I would like to apply this feature to a video feed then republish it to be recorded in a rosbag. 
Thank you.

Steve3nto, 2017-02-13:
I have the same issue. I think having the possibility to run the image_flip on a topic while maintaining all the other information untouched would be valuable. 

For example if I have a stereo camera mounted upside down I would have at least these topics..

camera/left/camera_info
camera/left/image_raw
...
camera/right/camera_info
camera/right/image_raw

Now I would like to have the data in image_raw flipped, while keeping the same timestamps for synchronization between left and right

A way to do that would be to call image_flip on the image_raw topics and republish them with only the data flipped but the same image_headers (sequence number, timestamp, encoding, frame_id...)

Have you thought about that possibility?

bbidault2, 2017-02-16:
@Steve3nto Are you capable to run the flip feature of '''image_publisher''' on a topic ? If so, how do you do it ? What is the command line ?
Regarding your question, correct me if I'm wrong but it seems like image_headers is untouched (in image_publisher_nodelet.cpp).

SteveMacenski, 2019-11-04:
Please ask questions that aren't bugs or feature requests on ROS Answers

closed 2019-11-04 by SteveMacenski