ros-perception/image_pipeline #264: Segmentation fault in every package of image pipeline

manuoso, 2017-04-06:
In first time, sorry for my english, it's not really good.

Well, when I try to run any package of image_pipeline, creating a catkin workspace and using catkin_make, there arent errors and it build good but give me a "segmentation fault (core dumped)" 

For example, I launch a two cameras using a driver: usb_cam and I want to view these images using a image_view: (in the catkin workspace) 
`rosrun` image_view image_view `image:=/mycamera/image_raw`
Later that using this command give me a **"segmentation fault (core dumped)"** It doesnt appear more info and the package image_view doesnt run.
I reviewed every dependence and every package that I need to use image_view is currently installed 

Finally If I install the package image_view using: sudo apt-get install ros-indigo-image-view and later I run the command: 
`rosrun` image_view image_view `image:=/mycamera/image_raw`
It works fine!!

I dont undestand why If I running the package of my own catkin workspace It isn't work...

I'm using ROS Indigo with Ubuntu 14.04 and I installed OpenCV 3.0.0.

k-okada, 2017-04-07:
Using opencv3 on Indigo is not recommended, consider using OpenCV 2 for Indigo or upgrade to Kinetic so that you can use OpenCV 3.0
If you still need using opencv3 on indigo, you can find hits at https://github.com/ros-perception/opencv_apps/issues/51#issuecomment-290444079.
But this will brings you another problem, so please cosider using opencv2/indigo or opencv3/kinetic.

manuoso, 2017-04-07:
And why it works fine when I installed the same package using:  
`sudo apt-get install `ros-indigo-image-view` ?
Its the same package and the same version of the package

k-okada, 2017-04-10:
that's because ros-indigo-image-view is compiled against opencv2, where as
you have compiled against opencv3 on your local environment, I suppose.

manuoso, 2017-04-10:
So thanks for all, I used opencv2 and now it works fine.
Fixed the issue, thanks

closed 2017-04-10 by manuoso