ros-perception/image_pipeline #265: Depth registration should interpolate when the depth image is smaller than the rgb image

Danmou, 2017-04-10:
As it is now, each pixel in the depth image is projected to one pixel in the registered image. This is a problem because the registered image has the same size as the color image, which is often larger than the depth image.

Example:
With the Intel Realsense R200 depth camera you could have a depth resolution of 320x240 and a color resolution of 640x480.
Here is a registered depth image of a wall (which has nearly complete coverage in the original depth image):
[image]
Because of the registration, only 1/4 of the pixels on the wall now have depth data, the rest are zero.
This causes problems with SLAM software such as RTABMAP, which will discard the visual features found in the color image if they do not have a valid depth associated with them, effectively causing 3/4 of the features to be discarded unnecessarily.

There is already a todo note in the [code](https://github.com/ros-perception/image_pipeline/blob/indigo/depth_image_proc/src/nodelets/register.cpp#L215):
`/// @todo When RGB is higher res, interpolate by rasterizing depth triangles onto the registered image`
although I am not sure what the proposed solution entails. Another option would be to scale the image before registration.

SteveMacenski, 2019-11-03:
PRs welcome, however scaling the image before registration would be a similar (if not exactly the same) process as interpolating point values between pixels to accomplish the same thing. 

This could be implemented as a nodelet chain of resize->registration, or could add the interpolation capability inside of registration itself.

machinekoder, 2020-02-12:
Thanks for the explanation. Is there more info on the `resize` operation? `image_proc` is sparsely documented.

lucasw, 2021-08-31:
Unless I'm missing something this is solved by the `fill_upsampling_holes` param from #363

mikeferguson, 2024-01-17:
Appears this is now supported by linked merged PR

closed 2024-01-17 by mikeferguson