ros-perception/image_pipeline #268: stereo_view show grey windows

antgarmun, 2017-04-12:
I use the command "rosrun image_view stereo_view stereo:=/ image:=image_raw" and I have the following output:

[ WARN] [1491997371.407180063]: There is a delay between when the camera drivers publish the raw images and when stereo_image_proc publishes the computed point cloud. stereo_view may fail to synchronize these topics without a large queue_size.
[ INFO] [1491997372.515601627]: Subscribing to:
	* /left/image_raw
	* /right/image_raw
	* /disparity
[ WARN] [1491997387.631961253]: [stereo_view] Low number of synchronized left/right/disparity triplets received.
Left images received:      419 (topic '/left/image_raw')
Right images received:     418 (topic '/right/image_raw')
Disparity images received: 0 (topic '/disparity')
Synchronized triplets: 0
Possible issues:
	* stereo_image_proc is not running.
	  Does `rosnode info /stereo_view_1491997371407059445` show any connections?
	* The cameras are not synchronized.
	  Try restarting stereo_view with parameter _approximate_sync:=True
	* The network is too slow. One or more images are dropped from each triplet.
	  Try restarting stereo_view, increasing parameter 'queue_size' (currently 5)
[ WARN] [1491997402.632357282]: [stereo_view] Low number of synchronized left/right/disparity triplets received.
Left images received:      836 (topic '/left/image_raw')
Right images received:     835 (topic '/right/image_raw')
Disparity images received: 0 (topic '/disparity')
Synchronized triplets: 0
Possible issues:
	* stereo_image_proc is not running.
	  Does `rosnode info /stereo_view_1491997371407059445` show any connections?
	* The cameras are not synchronized.
	  Try restarting stereo_view with parameter _approximate_sync:=True
	* The network is too slow. One or more images are dropped from each triplet.
	  Try restarting stereo_view, increasing parameter 'queue_size' (currently 5)
[ WARN] [1491997417.632702610]: [stereo_view] Low number of synchronized left/right/disparity triplets received.
Left images received:      1261 (topic '/left/image_raw')
Right images received:     1261 (topic '/right/image_raw')
Disparity images received: 0 (topic '/disparity')
Synchronized triplets: 0
Possible issues:
	* stereo_image_proc is not running.
	  Does `rosnode info /stereo_view_1491997371407059445` show any connections?
	* The cameras are not synchronized.
	  Try restarting stereo_view with parameter _approximate_sync:=True
	* The network is too slow. One or more images are dropped from each triplet.
	  Try restarting stereo_view, increasing parameter 'queue_size' (currently 5)
[ WARN] [1491997432.632364353]: [stereo_view] Low number of synchronized left/right/disparity triplets received.
Left images received:      1689 (topic '/left/image_raw')
Right images received:     1689 (topic '/right/image_raw')
Disparity images received: 0 (topic '/disparity')
Synchronized triplets: 0
Possible issues:
	* stereo_image_proc is not running.
	  Does `rosnode info /stereo_view_1491997371407059445` show any connections?
	* The cameras are not synchronized.
	  Try restarting stereo_view with parameter _approximate_sync:=True
	* The network is too slow. One or more images are dropped from each triplet.
	  Try restarting stereo_view, increasing parameter 'queue_size' (currently 5)

I know that the subscribed topics are correct becouse I have checked with rostopic list. I don't know what is the problem. The output sais: "stereo_image_proc is not runnig", so I tried: "rosrun stereo_image_proc stereo_image_proc" and It gives me the following output:

[ WARN] [1491997643.046852891]: Started in the global namespace! This is probably wrong. Start stereo_image_proc in the stereo namespace.
Example command-line usage:
	$ ROS_NAMESPACE=my_stereo rosrun stereo_image_proc stereo_image_proc

The output talks about a problem with ROS_NAMESPACE, but in my case, ROS_NAMESPACE is empty. What can I do?

k-okada, 2017-04-13:
did you try
```
* The cameras are not synchronized.
Try restarting stereo_view with parameter _approximate_sync:=True
```

antgarmun, 2017-04-17:
@k-okada I used the command "rosrun image_view stereo_view stereo:=/ image:=image_raw _approximate_sync:=True" and I had the following output:

kwduser@hiroconsole:~$ rosrun image_view stereo_view stereo:=/ image:=image_raw _approximate_sync:=True
[ WARN] [1492412435.724658146]: There is a delay between when the camera drivers publish the raw images and when stereo_image_proc publishes the computed point cloud. stereo_view may fail to synchronize these topics without a large queue_size.
[ INFO] [1492412436.794372983]: Subscribing to:
	* /left/image_raw
	* /right/image_raw
	* /disparity
[ WARN] [1492412451.902390282]: [stereo_view] Low number of synchronized left/right/disparity triplets received.
Left images received:      414 (topic '/left/image_raw')
Right images received:     414 (topic '/right/image_raw')
Disparity images received: 0 (topic '/disparity')
Synchronized triplets: 0
Possible issues:
	* stereo_image_proc is not running.
	  Does `rosnode info /stereo_view_1492412435722359189` show any connections?
	* The cameras are not synchronized.
	  Try restarting stereo_view with parameter _approximate_sync:=True
	* The network is too slow. One or more images are dropped from each triplet.
	  Try restarting stereo_view, increasing parameter 'queue_size' (currently 5)

Stereo view still shows grey windows.

k-okada, 2017-04-21:
```
Disparity images received: 0 (topic '/disparity')
```
did you start `stare_proc` to publish disparity images?

antgarmun, 2017-04-26:
@k-okada, I dind't start stare_proc, how can I do it? Where can I find information about it?

k-okada, 2017-04-27:
you can learn how to use stereo_image_proc at ->
http://wiki.ros.org/stereo_image_proc/Tutorials/ChoosingGoodStereoParameters

antgarmun, 2017-05-25:
@k-okada I have started stereo_image proc but I only get tree grey windows with stereo view. I run the command: "rosrun image_view stereo_view stereo:=/ image:=image_mono _aproximate_sync:=True" and I get the following output:
[ INFO] [1495711913.321762184]: Subscribing to:
	* /left/image_mono
	* /right/image_mono
	* /disparity
[ WARN] [1495711928.422489282]: [stereo_view] Low number of synchronized left/right/disparity triplets received.
Left images received:      833 (topic '/left/image_mono')
Right images received:     838 (topic '/right/image_mono')
Disparity images received: 0 (topic '/disparity')
Synchronized triplets: 0
Possible issues:
	* stereo_image_proc is not running.
	  Does `rosnode info /stereo_view_1495711912251167985` show any connections?
	* The cameras are not synchronized.
	  Try restarting stereo_view with parameter _approximate_sync:=True
	* The network is too slow. One or more images are dropped from each triplet.
	  Try restarting stereo_view, increasing parameter 'queue_size' (currently 5)

I have used the command "rosnode info /stereo_view_1495711912251167985" and I got the following output:
rosnode info /stereo_view_1495711912251167985
--------------------------------------------------------------------------------
Node [/stereo_view_1495711912251167985]
Publications: 
 * /rosout [rosgraph_msgs/Log]

Subscriptions: 
 * /disparity [stereo_msgs/DisparityImage]
 * /left/image_mono [sensor_msgs/Image]
 * /right/image_mono [sensor_msgs/Image]

Services: 
 * /stereo_view_1495711912251167985/get_loggers
 * /stereo_view_1495711912251167985/set_logger_level

contacting node http://hiroconsole:42946/ ...
Pid: 31585
Connections:
 * topic: /rosout
    * to: /rosout
    * direction: outbound
    * transport: TCPROS
 * topic: /left/image_mono
    * to: /left/image_proc (http://hiroconsole:45128/)
    * direction: inbound
    * transport: TCPROS
 * topic: /left/image_mono
    * to: /stereo_image_proc (http://hiroconsole:60879/)
    * direction: inbound
    * transport: TCPROS
 * topic: /right/image_mono
    * to: /right/image_proc (http://hiroconsole:41251/)
    * direction: inbound
    * transport: TCPROS
 * topic: /right/image_mono
    * to: /stereo_image_proc (http://hiroconsole:60879/)
    * direction: inbound
    * transport: TCPROS
 * topic: /disparity
    * to: /stereo_image_proc (http://hiroconsole:60879/)
    * direction: inbound
    * transport: TCPROS

So, I supposed that the stereo_image_proc is running becouse of the output.

k-okada, 2018-05-19:
@antgarmun I have updated launch files and it will help you to use approximate sync features more easily
https://github.com/ros-perception/image_pipeline/pull/338
You may also better to see similar issue thread https://github.com/ros-perception/image_pipeline/issues/323

JWhitleyAStuff, 2019-04-25:
Closing due to lack of response and also merging of #338.

closed 2019-04-25 by JWhitleyAStuff